convertersystems/opc-ua-client #2: Refreshing monitored items on reconnection of device

josetruyol-axys, 2016-08-25:
Hi,

I'm writting a control software / HMI using this ua-client for WPF, with Kepwere as OPC Server.

Today I was doing some connection tests, disconnecting the ethernet cable of my PLC and change some monitored item value in the software. Obviously this change wont take effect because the connection state, but, after reconnect why don't refresh the PLC tag with the value that I previously set?

Or this is the expected behavior and I need to make some internal storage variable to replicate this?
Is there some event that notify when a suscription (in this case, the PLC) is no responding?

Best Regards
Jose Truyol

josetruyol-axys, 2016-08-25:
Kepserver have a System.NoError Tag that indicate if there is an error on the PLC, so that part is already done. But I'm trying to resend the latest change while was offline.

I overwrote the SetProperty method:

``` 
protected override bool SetProperty<T>(ref T storage, T value, [CallerMemberName] string propertyName = null)
{
            
    if(ServerState != CommunicationState.Opened || (!System_NoError && "System_NoError" != propertyName))
    {
        m_eventsPending.Add(new WriteEvent
        {
            PropertyName = propertyName,
            PropertyValue = value
        });
        return false;
    }
            
    return base.SetProperty(ref storage, value, propertyName);
}
  ```
And then listen to the server state change event with this:
```
public void OnStateChangeEvent(CommunicationState state)
{
    ServerState = state;
}
```
On every change on System_NoError I do:
  ```
        private void OnError()
        {
            if(System_NoError)
            {
                while(m_eventsPending.Count > 0)
                {
                    WriteEvent e = m_eventsPending.ElementAt(0);
                    bool storage = !(bool)e.PropertyValue;
                    var p = this.GetType().GetProperty(e.PropertyName);
                    p.SetValue(this, storage);
                    p.SetValue(this, e.PropertyValue);
                    m_eventsPending.RemoveAt(0);
                }
            }
        }

  ```

I do some initialization and keep it on the m_eventsPending list, and start sending when connection is complete (Opened), but nothing happend on the PLC side. What could be wrong?

awcullen, 2016-08-26:
Hi Jose,
Thank you for reporting this issue!

I was working on a lime kiln hmi recently. The operator can start/stop conveyors, set the setpoints of the variable speed drives, etc.  If the communication failed, the client wanted the hmi to display an overlay that disabled the controls and show a message like "Connection Lost!".  In this case, the client did not want any conveyors to start if the hmi reconnected.

In your case, I understand you wish the Write method to wait until "System_NoError" is true.

(Maybe the Write should repeat until it completes successfully or times out after x seconds?)

Could you tell us what error message is returned from the write when the cable is unplugged? It will be in the debug console like "Subscription error writing value for NodeId {nodeId}. {ex.Message}"

BR,
Andrew

josetruyol-axys, 2016-08-26:
Hi Andrew,
Thanks you for your reply.

The error message returned when the cable is unplugged is:
```
SemaforosNNM.vshost.exe Warning: 0 : Subscription error writing value for NodeId ns=2;s=ModbusDelta.KM0.M Registers.M119. Timeout occurred while processing the request.
```

I notice that, for every tag, the ua client try to write the value 2 times only, and then forget about it, even on replug . If I unplug the cable and replug after a few seconds (less than 10), the value is correctly write.

If I wait 2 o 3 minutes every property associated to this PLC (subscription) is reset to the default value (in my case, all to false), raising the PropertyChanged event, even when this not really happend in the real world.

Best regarts,
Jose

awcullen, 2016-08-28:
Would you consider directly writing the tags in your code? For example, this code writes the green, yellow, and red traffic lights when triggered by a property change, repeating every five seconds, until successful.

```
public class LightsViewModel : SubscriptionBase
{        
        private bool green;
        private bool red;
        private bool yellow;
        private CancellationTokenSource cts;

        public bool Green
        {
            get { return this.green; }
            set
            {
                this.green = value;
                this.WriteOutputsAsync();
            }
        }
        // add yellow and red

        private async void WriteOutputsAsync()
        {
            var request = new WriteRequest
            {
                NodesToWrite = new[]
                {
                    new WriteValue
                    {
                        NodeId = NodeId.Parse("ns=2;s=GreenLight"),
                        AttributeId = AttributeIds.Value,
                        Value = new DataValue(this.green)
                    },
                    new WriteValue
                    {
                        NodeId = NodeId.Parse("ns=2;s=YellowLight"),
                        AttributeId = AttributeIds.Value,
                        Value = new DataValue(this.yellow)
                    },
                    new WriteValue
                    {
                        NodeId = NodeId.Parse("ns=2;s=RedLight"),
                        AttributeId = AttributeIds.Value,
                        Value = new DataValue(this.red)
                    }
                }
            };

            var cts = new CancellationTokenSource();
            var cts1 = Interlocked.Exchange(ref this.cts, cts);
            cts1?.Cancel();

            try
            {
                while (!await this.TryWriteAsync(request))
                {
                    await Task.Delay(5000, cts.Token);
                }
            }
            catch (TaskCanceledException)
            {
            }
        }

        private async Task<bool> TryWriteAsync(WriteRequest request)
        {
            try
            {
                var response = await this.Session.WriteAsync(request);
                for (int i = 0; i < response.Results.Length; i++)
                {
                    if (StatusCode.IsBad(response.Results[i]))
                    {
                        Debug.WriteLine($"Error writing output '{request.NodesToWrite[i].NodeId}' : {StatusCodes.GetDefaultMessage(response.Results[i])}");
                        return false;
                    }
                }

                return true;
            }
            catch (Exception ex)
            {
                Debug.WriteLine($"Error writing outputs : {ex.Message}");
                return false;
            }
        }
}
```

josetruyol-axys, 2016-08-31:
Hi, Andrew.

Today I will continue with my project. I have implemented your suggested methods... So far, so good. 

I'm a novice C# programmer, and don't know very much about many advanced language feature. So I read about WeakReference and a question arose.

If your are implementing WeakReference in the subscription, this means that if the PLC don't respond after awhile, the garbage collector will delete the subscription until PLC reconnection?

I make this question because after some time, after unplug the PLC, each Property return to their default values, and affect the background logic.

Best Regarts,
Jose

awcullen, 2016-08-31:
I have a theory, would you test this for me?

All Properties returning to False is probably caused by Kepware server sending a PublishResponse with all False values.  The server would do this after initial Connect and maybe after Reconnect. I believe the UaTcpSessionClient is reconnecting every 2 minutes due to the errors and the Kepware server is responding by publishing all false values. You may be able to modify this behavior.

First, you can change the 2 minute timeout to delay the reconnection. When you construct the UaTcpSessionClient, change the sessionTimeout  to 5 minutes like this.
```
    var session = new UaTcpSessionClient(appDescription, appCertificate, userIdentity, endpointUrl, sessionTimeout: 5 * 60000); 
```

Second, you may be able to change the behavior of Kepware to return the last known value of a tag after reconnecting.  Find the Device Properties page. Try checking the box 'Provide initial updates from cache'.  I'm just guessing here.

[image]

Andrew

awcullen, 2016-10-10:
Hopefully this issue is solved. Feel free to reopen if there is still a problem.

closed 2016-10-10 by awcullen