convertersystems/opc-ua-client #3: Server state change event & timeout events

josetruyol-axys, 2016-08-26:
Hi Andrew,

I noticed that there is no way to add a event listerner to the ServerStateChange.
Digging on your code I saw that is have not access to SuscriptionAdapter because the

awcullen, 2016-08-29:
Hello Jose,
The reason for having no direct way to add an event handler for session's StateChanged event is that we wanted to use weak references. In WPF and UAP, the views and viewModels are usually just abandoned by the shell when closed or navigated away.  The abandoned viewModels are garbage collected. When the Session detects this, it deletes the Subscription on the Server.  If I provide a public event on Session for StateChanged, it would be easy to add a handler in the viewModel and forget to remove it.  

Regarding write timeouts, I upgraded the UaClient  and Samples to enable a feature where the latest error detected when creating, publishing and writing the MonitoredItems are available in the Subscription.
In WPF, the error is displayed the same way validation errors are displayed. In UAP, there is no display.

[image]

Let me know if this solves your issue. I appreciate your feedback.

Andrew

josetruyol-axys, 2016-08-31:
>  ... the views and viewModels are usually just abandoned by the shell when closed or navigated away ...

In this case, I will use only one view, without any kind of navigation. I guess that this will be not a issue, or I'm wrong?

>  I upgraded the UaClient and Samples to enable a feature where the latest error detected when creating,

I updated my project with your last commit, and I noticed the change on the ` ISubscription ` interface. Thanks for that!

Best Regarts,
José

awcullen, 2016-10-10:
Hopefully this issue is solved. Feel free to reopen if there is still a problem.

closed 2016-10-10 by awcullen