convertersystems/opc-ua-client #5: GetEndpointAsync returns endpoints pointing to "testopcua" host??

basprins, 2016-10-28:
I just installed "Workstation.UaClient" via NuGet, and I can't seem to figure out who or what is causing the issue that all found endpoint urls resolve to "opc.tcp://testopcua:4840". 

I am discovering endpoints on: opc.tcp://localhost.4840.

Any idea what this means? Is this some kind of defaulted host name? 

If it helps, this is all I have, and this results in the testopcua hostname in the endpoint url:

            var discoveryUrl = "opc.tcp://localhost:4840";

            var getEndpointsRequest = new GetEndpointsRequest
            {
                EndpointUrl = discoveryUrl,
                ProfileUris = new[] { TransportProfileUris.UaTcpTransport }
            };

            var result = await UaTcpDiscoveryClient.GetEndpointsAsync(getEndpointsRequest);

            // result.Endpoints now contain a three endpoints all pointing to "testopcua"

awcullen, 2016-10-28:
Looks like the server at 4840 is having trouble getting the name of your computer. 
Maybe you can investigate with another free OPC UA Client. For instance, UA Expert from Unified Automation.  https://www.unified-automation.com/downloads/opc-ua-clients.html

basprins, 2016-10-29:
Dear Andrew, 

You are correct. The hostname was set by B&R OPC UA server. The weird thing is, is that I did not experience this problem with other OPC UA clients I tried (e.g. UA expert...). But that's alright, I thought I missed something silly but I can solve it properly now by setting the correct hostname in B&Rs OPC UA server.

Thanks for answering and pointing me in the right direction! Much appreciated.

Best regards, Bas

basprins, 2016-10-29:
And then I started to actually read what that UA expert dialog was trying to tell me every time I hit "yes"....

[image]

So your suggestion is hereby confirmed I guess :)

closed 2016-10-29 by basprins