convertersystems/opc-ua-client #6: Read/Write parameters using identifier

basprins, 2016-10-29:
Dear Andrew,

Is your library able to retrieve parameters using the OPC UA namespace Identifier? 

[image]

Would you have any documentation on this? In fact this is all I really need to know; how to read/write parameter values over OPC UA using this library. Until now I don't really have the helicopter view of what I should do (using your examples as a starting point).

                        var subscriptionResponse = await session.CreateSubscriptionAsync(subscriptionRequest);
                        var id = subscriptionResponse.SubscriptionId;

                        // Try to read value ::AsGlobalPV:InputUnit.Command

                        var itemsToCreate = new[]
                        {
                            new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2258"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }
                        };
                        var itemsRequest = new CreateMonitoredItemsRequest
                        {
                            SubscriptionId = id,
                            ItemsToCreate = itemsToCreate,
                        };
                        var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);

PS: the "i=2258" is copy pasted from your example, no idea what it refers to. I was kind of hoping to be able to replace it with "::AsGlobalPV:InputUnit.Command" but that wont work.

Any advice/help/pointers would be greatly appreciated!

awcullen, 2016-10-29:
Node Id's can be integer, string, guid, or binary.  NodeId.Parse expects a string formatted with:
ns=<namespace index>; [ or nsu=<namespace url>; ] [ or nothing for index = 0 ]
i=<integer id> [ or s=<string id> ] [ or g=<guid id> ] or [ b=<binary id> ]
Try this:
"ns=6;s=::AsGlobalPV:InputUnit.Command"

basprins, 2016-10-29:
Hi Andrew,

Works like a charm. Nice!

                        var readResponse = await session.ReadAsync(
                            new ReadRequest
                            {
                                NodesToRead = new []
                                {
                                    new ReadValueId
                                    {
                                        NodeId = NodeId.Parse("ns=6;s=::AsGlobalPV:InputUnit.Configuration.AirKnifeOffHeightBelowPick"),
                                        AttributeId = AttributeIds.Value
                                    }
                                }
                            });

                        // readResponse contains current value

                        var writeResponse = await session.WriteAsync(
                            new WriteRequest
                            {
                                NodesToWrite = new[]
                                {
                                    new WriteValue
                                    {
                                        NodeId = NodeId.Parse("ns=6;s=::AsGlobalPV:InputUnit.Configuration.AirKnifeOffHeightBelowPick"),
                                        AttributeId = AttributeIds.Value,
                                        Value = new DataValue((double)20),
                                        IndexRange = null,
                                    }
                                }
                            });

                        // value is correctly updated in PLC

One last question, can you briefly explain what I can expect from subscriptions? Does it mean I actually get changed events when the OPC UA property in the server is updated?

awcullen, 2016-10-29:
There's two ways to implement subscriptions with this package.

First, there's the low-level api you can use. This involves calling PublishAsync in a loop and dealing with the response. Please see sample here; https://github.com/convertersystems/opc-ua-samples/blob/master/ConsoleApp/Program.cs 

Second, there's a higher level api for WPF, UWP, Xamarin apps. This involves creating a "view model" class that implements INotifyPropertyChanged and with properties for each node you wish to monitor. An example of such a class is here: https://github.com/convertersystems/opc-ua-samples/blob/master/StatusHmi/MainViewModel.cs
You create a live "view model" subscription like so:
`
var viewModel = session.CreateSubscription<MainViewModel>();
`

Then you create a "view" with bindings to the "view model". 

A sample using WPF is here: https://github.com/convertersystems/opc-ua-samples/tree/master/StatusHmi

basprins, 2016-10-29:
Thanks for the answer. I had already peeked in those examples. And yes, the MVVM approach definitely sounds interesting, but we also have a "non UI related" use for subscriptions / events. So I guess we'll at least go for the low level variant.
 
But still, the reason for my question was, that I was a bit surprised to see the while-loop implementation:

                while (!Console.KeyAvailable)
                {
                    var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

                    // loop thru all the data change notifications
                    var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                    foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            Console.WriteLine($"clientHandle: {min.ClientHandle}; value: {min.Value}");
                        }
                    }

                    publishRequest = new PublishRequest
                    {
                        SubscriptionAcknowledgements = new[] { new SubscriptionAcknowledgement { SequenceNumber = publishResponse.NotificationMessage.SequenceNumber, SubscriptionId = publishResponse.SubscriptionId } }
                    };
                }

That kind of struck me as "polling". I thought OPC UA was event driven by nature. I kind of expected to receive changed event notifications on socket level. Am I misunderstanding the concept in OPC UA? Isn't the server _sending_ an update to the client if there are subscriptions set?

PS: just 100% curiosity, but there are a lot of similarities in the .NET implementations of OPC UA clients. The commercial version (unifiedautomation.com), the OPCfoundation/UA.NET, and your implementation. Are you guys forking stuff from each other or is it completely separate development? If the latter, did you really write this complete stack _on your own_?! That seems a rather huge job for one person.

Again, many thanks for your time and help!

basprins, 2016-10-31:
Hi Andrew,

I also have the subscriptions working and will build an event mechanism on top of your library (a rather trivial change). 

I still don't fully understand what happens under the hood PublishAsync. If you can find the time to elaborate then that would be much appreciated. I doesn't seem that you are polling for value changes by actually retrieving server side values. But on the other hand I can't seem to understand / follow what _does_ happen under the hood.

Anyway, thanks a bunch for helping me out. As things stand now we can (and will) move forward with your library (and can get rid of a payed version!). 

Cheers, Bas

awcullen, 2016-10-31:
The UA specification chose polling as the mechanism for getting data changes and events from subscriptions. But it's not as bad as I seems. First, this is long polling, meaning the server holds the PublishRequest until it has data to send (or a keep-alive). Second, the client can send multiple PublishRequests to allow the server to respond in bursts if many subscriptions change at once. Third, the client fills in each PublishRequest with the id's of previous received PublishResponses, letting the server know the subscription is still alive.

Will this be changed in the next version of the Spec, I don't know

awcullen, 2016-10-31:
If you wish to share your event mechanism, I invite you to fork the project and push your work in progress. It's super easy, I would be happy to help out.

basprins, 2016-11-03:
Hi Andrew,

Sorry for the late response, something else pulled me away from this topic. But that explained a lot. Long pulling icw keep alive ticks. So as you said, it's not as "bad" as I thought it was (not that it would be a huge issue in the first place, I mostly just like to have a bit of understanding of what's going on).

I don't mind to fork when we get to it. As you said, a rather trivial change (like two lines of code to add the event handler, and fire it in the for loop where all changed monitored items are iterated). 

I will open a new ticket with a new question, so that you can close this one. If the next question is not suitable for this forum, then it will not pollute this Q&A (as this one is more than answered).

Many thanks! 
Bas

closed 2016-11-03 by basprins