convertersystems/opc-ua-client #7: 1 subscription <> 1 monitored item  vs 1 subscription <> many monitored items

basprins, 2016-11-03:
Hi Andrew,

Maybe the wrong place for this question, if so I hope you can tell me where to learn more.

We currently use unifiedautomation for communicating with our B&R OPC UA server. There we implemented the subscriptions by the design of 1 monitored item per subscription (1 to 1 relation). No specific reason, really, just a convenience and we found out it worked fine this way so didn't bother looking in to proper guidelines for this.

Now I injected your library as a replacement of the existing OPC UA client. It seems to work fine in a prototype solution:
- I can read a property value
- I can write a property value
- I can subscribe to changed values

However, when I start our application on top of your OPC UA client, it complains about too many publish requests active on the server.

> The server has reached the maximum number of queued publish requests.

When I change the initial design so that I have a 1 to many relation (= 1 sub => roughly 80-100 monitored items), the OPC UA communication slows down the software drastically. I haven't measured, but in order to receive updates for those 80-100 items it seems to stress the CPU and takes seconds to synchronize.

Since my knowledge wrt OPC UA is still limited, I also don't know which settings I should change in order to improve this behavior. When I google for this problem I end up on forums for java or python related APIs, but again, my limited knowledge of OPC UA doesn't help to understand how to solve this. 

What am I doing wrong that causes this to happen with your library while the same approach works fine with another library? Can you advise me on which settings I should focus on / change? I have checked in the configuration of the server (B&R PLC) but it doesn't allow me anything meaningful for subscriptions. 

Just a few snippets of how things are implemented now (if it helps):

The most relevant class for this topic, the '**OpcuaProperty**'

```
    public class OpcuaProperty<T>
    {
        private readonly UaTcpSessionChannel session;
        private readonly NodeId nodeId;

        public event EventHandler ValueChanged;

        private T value;
        public T Value
        {
            get { return value; }
            set
            {
                this.value = value;
                session.WriteAsync(
                    new WriteRequest
                    {
                        NodesToWrite = new[]
                        {
                                    new WriteValue
                                    {
                                        NodeId = nodeId,
                                        AttributeId = AttributeIds.Value,
                                        Value = new DataValue((T)value),
                                        IndexRange = null,
                                    }
                        }
                    }).Wait();
            }
        }

        public OpcuaProperty(
            UaTcpSessionChannel session,
            NodeId nodeId)
        {
            this.session = session;
            this.nodeId = nodeId;

            var subscriptionRequest = new CreateSubscriptionRequest
            {
                RequestedPublishingInterval = 1000,
                RequestedMaxKeepAliveCount = 1,
                RequestedLifetimeCount = 30,
                PublishingEnabled = true
            };

            var subscriptionResponse = session.CreateSubscriptionAsync(subscriptionRequest).Result;
            var id = subscriptionResponse.SubscriptionId;

            var itemsToCreate = new[]
            {
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor = new ReadValueId
                    {
                        NodeId = nodeId,
                        AttributeId = AttributeIds.Value
                    },
                    MonitoringMode = MonitoringMode.Reporting,
                    RequestedParameters =
                        new MonitoringParameters
                        {
                            ClientHandle = 12345,
                            SamplingInterval = -1,
                            QueueSize = 0,
                            DiscardOldest = true
                        }
                }
            };

            var itemsRequest = new CreateMonitoredItemsRequest
            {
                SubscriptionId = id,
                ItemsToCreate = itemsToCreate,
            };
            session.CreateMonitoredItemsAsync(itemsRequest).Wait();

            Task.Run(() => Publish());
        }

        private async void Publish()
        {
            var publishRequest = new PublishRequest
            {
                SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
            };

            while (true)
            {
                var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

                // loop thru all the data change notifications
                var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                foreach (var dcn in dcns)
                {
                    foreach (var min in dcn.MonitoredItems)
                    {
                        ValueChanged?.Invoke(this, EventArgs.Empty);
                    }
                }

                publishRequest = new PublishRequest
                {
                    SubscriptionAcknowledgements = new[] { new SubscriptionAcknowledgement { SequenceNumber = publishResponse.NotificationMessage.SequenceNumber, SubscriptionId = publishResponse.SubscriptionId } }
                };
            }
            // ReSharper disable once FunctionNeverReturns
        }
    }
```

The session wrapper (doesn't do much, but here it is anyway):

```
    public class OpcuaControl : IOpcuaControl
    {
        private UaTcpSessionChannel session;
        private static ushort NamespaceId = 6; // TODO: Hardcoded for now...

        public async Task Connect(
            string hostname,
            string port)
        {
            var discoveryUrl = $"opc.tcp://{hostname}:{port}"; 
            
            var getEndpointsRequest = new GetEndpointsRequest
            {
                EndpointUrl = discoveryUrl,
                ProfileUris = new[] {TransportProfileUris.UaTcpTransport}
            };

            var result = await UaTcpDiscoveryClient.GetEndpointsAsync(getEndpointsRequest);

            if (result.Endpoints == null || result.Endpoints.Length == 0)
            {
                throw new Exception(); // todo throw specific
            }

            var endPoint = result.Endpoints.OrderBy(e => e.SecurityLevel).First();

            var appDescription = new ApplicationDescription
            {
                ApplicationName = "bla",
                ApplicationUri = "urn:127.0.0.1:4840",
                ApplicationType = ApplicationType.Client
            };

            session = new UaTcpSessionChannel(
                appDescription,
                appDescription.GetCertificate(),
                new AnonymousIdentity(),
                endPoint);

            await session.OpenAsync();
        }

        public OpcuaProperty<T> CreateOpcuaProperty<T>(string iid)
        {
            return new OpcuaProperty<T>(session, new NodeId(iid, NamespaceId));
        }

        public void IncrementLifeSign()
        {
            // todo
        }
    }
```

awcullen, 2016-11-04:
Hi Bas,
Here's a couple of things to look at, while I put together a sample for this.

I think most libraries send a PublishRequest for each subscription that's been created. That's not a rule. A PublishRequest is not directed to a specific subscription. You can have 1 to any number queued up at the server. But as you've seen, the server has a mechanism to say that it's queue is full. It returns a response that raises a ServiceResultException ( with HResult =0x80780000u //  BadTooManyPublishRequests).

You may have to reconsider having 100 subscriptions ( each with 1 MonitoredItem), only because the other libraries may be queuing up way too may PublishRequests. I've often used subscriptions with hundreds of items. I usually get better performance, because there is less overhead.

But, I would change RequestedMaxKeepAliveCount to 10 or more. 

Andrew

basprins, 2016-11-04:
Hi Andrew,

That again was a very useful comment. I still have the 1 sub for many monitored items, but as you suggested the problem was the PublishAsync per subscription. Once I got rid of that, I was able to run our software stack on top of it without any issues! 

Thank you very much for your help! I hope I can bug you again when we need help!

Cheers, 

Bas

awcullen, 2016-11-09:
Reopen if you need to.

closed 2016-11-09 by awcullen

srohit1989, 2022-11-10:
Hello Bas,

I did the same way that you tried in above example but still getting data from kepserver almost after 50sec