convertersystems/opc-ua-client #8: Timeout on subscription during debugging

basprins, 2016-11-07:
Hi Andrew,

Hopefully an easy one to answer. I am running into "subscription timeout exceptions" during debugging. 

I tried to set the "timeout hints" on RequestHeaders (ridiculously high) for the subscription creating and the PublishAsync request, but that has no effect. I also set the timeout on the session ridiculously high but that also has no effect.

Is it possible to set a timeout so that I can debug as slow as I desire without running into timeout exceptions?

Thanks

awcullen, 2016-11-07:
var subscriptionRequest = new CreateSubscriptionRequest
                {
                    RequestedPublishingInterval = 1000, // 1 sec
                    RequestedMaxKeepAliveCount = 10, // 10 sec
                    RequestedLifetimeCount = 3600, // 1 hour
                    PublishingEnabled = true
                };

awcullen, 2016-11-08:
Please note, when calling PublishAsync you should set the timeoutHint to a long value, say 120000 (2 minutes). This is because the PublishRequest is purposefully held in a queue on the server for some time.

basprins, 2016-11-08:
Hmm, it keeps throwing "There is no subscription available for this session".

I also tried a longer timeout hint for the PublishAsync (since the exception is thrown on that method) but that doesn't seem to matter.

public const uint BadNoSubscription = 0x80790000u; (= There is no subscription available for this session.)

There are no references to `BadNoSubscription`. Does that mean it's generated by the server completely? E.g. `StatusCodes.BadRequestTimeout` is referred when `CancelRequest` is called. So in order to get rid of this "No subscription for session" error I should be able to tell the server to relax the timeouts, right? That's done with "hints" I guess? Are there more timeouts hints (apart from the hint for the PublishRequest) I should set?

awcullen, 2016-11-08:
Yes, this error is generated by the server when you call PublishAsync and there are no subscriptions. The way I handle it is to create an empty subscription after I open the connection. A subscription with no monitored items will receive an empty publishResponse at an interval of PublishingInterval * KeepAliveCount.

basprins, 2016-11-09:
OK, that gave the insights I needed. Thanks for you help. It's working now. I can debug as long as I like without running into timeouts. Both the PublishAsync (timeout hint) and the properties you described on the subscription request did the trick after all.

closed 2016-11-09 by basprins