convertersystems/opc-ua-client #9: Net Standart Package installing uneeded dependecies in NET 4.6 Project

jogibear9988, 2016-12-11:
I'm using Net 4.6, but all the Net Standart Packages, like System.Filesystem, System.Appcontext,... are installed if I'm using the Nuget

awcullen, 2016-12-13:
I took a look at System.IO.FileSystem.dll in the bin directory of my 4.6 sample app. First, I noticed the file size is only 23 K. I opened the dll in the tool ILSpy, and saw that it only contains attributes that forward the type loader to mscorlib, 4.0.
So I concluded that it is needed to enable portability to all the platforms I would like to support.

jogibear9988, 2016-12-13:
okay, seems they onlynot needed if you create a 4.5 or 4.6 version directly

closed 2016-12-14 by awcullen

awcullen, 2016-12-18:
@jogibear9988,
Workstation.UaClient 1.5.1 supports .NET 4.5. There are fewer references as you suggested.