convertersystems/opc-ua-client #11: Visual Studio 2015 support

abrasat, 2017-01-05:
Could you please add also a VS2015 project file for the opc-ua-client ? The project cannot actually be loaded in VS2015.

abrasat, 2017-01-06:
I installed VS2017 RC, but still cannot load opc-ua client project. The error message says that file "C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Sdks\Microsoft.NET.Sdk\Sdk\Sdk.props" cannot be found.

Wher does this file come from ?

awcullen, 2017-01-08:
I chose to install VS2017 Community with the following Workloads:
Universal WIndows Platform development.
.NET Desktop development.
Node.js development.

bangonkali, 2017-01-10:
Is there any specific reason why this has to be on VS2017 at this point?

Tbh, i also worked on Xamarin and it seems the VS2017 support for both iOS and Android development were personally what worked for me. However VS2017 is still in the bleeding edge.

awcullen, 2017-01-11:
I wanted to use the multi-target feature to support both NetStandard and Net45. I tried a Shared Project, but my TextTemplating stopped working. If anyone needs a VS2015 solution, let me know.

awcullen, 2017-01-18:
Close for now.

abrasat, 2017-01-19:
Both VS2017 RC and VS2015 (I have them both installed on my PC)  fail to open the OPC UA client project. The VS2017 RC pops up the error I described above (... sdk.props cannot be found)
In the UaClient.csproj file the first line is:
<Project Sdk="Microsoft.NET.Sdk" ToolsVersion="15.0">

Where does the Microsoft.NET.Sdk come from ?
What  .NET Core versions are installed in your development environment ?

awcullen, 2017-01-19:
I recently setup VS2017RC on a new machine and found I needed to install these four 'Workloads" to open both the client and samples repos:
Universal Windows Platform development.
.NET Desktop development.
Node.js development.
.NET Core (and Docker)

If this doesn't work for you, I'll make a VS2015 project. Let me know if you prefer the Net45 or NetStandard target.

bangonkali, 2017-01-20:
Hi @awcullen it would be great if we have NetStandard targets. 

Could you point to a good resource on how to easily make nugets or preferably projects within the same solution using VS2017 that can work with Xamarin PCL (Forms) and NetStandard CLI, is this possible?

abrasat, 2017-01-20:
An additional VS2015 Project with NetStandard target would be a good option.

awcullen, 2017-01-20:
@bangonkali I recommend following @onovotny. His projects and blog demonstrate how VS2017 simplifies building multi-targeting libraries. 

@abrasat  I hand edited it for now. The Property Pages in 2017RC are a little messed up.

To both, I am uploading the VS2015 solution later today.

awcullen, 2017-01-20:
I think I created a new multi target project and used that as template.

All: I uploaded the VS2015 solution.  I will close this issue for now.
p.s.  I was having trouble loading VS2017RC and Siemens TIA Portal on the same computer for an upcoming business trip. So I had to return to VS2015 anyway.

closed 2017-01-20 by awcullen