convertersystems/opc-ua-client #12: Getting the NamespaceIndex right?

WoodsterDK, 2017-01-06:
Hi There,

I'm connecting to a Kepware OPC Server, and am wondering if there is an easy way to get the right NameSpaceIndex used for creating the 'NodeId's for the 'MonitoredItemCreateRequest' - the number 2 is the right for my instance, but will this number be static?

awcullen, 2017-01-06:
I think the Kepware folks know all hell would break loose if the NameSpaceIndex changed.
But you can use this:
```
var n = ExpandedNodeId.ToNodeId(ExpandedNodeId.Parse("nsu=http://opcfoundation.org/UA/;i=2256"), this.session.NamespaceUris);
```

WoodsterDK, 2017-01-09:
Thanks... My knowledge about the protocol/structure is in the start phase.
Will give it a shot.
Do you by the way have any links to some nice to know documentation about the OPC?
Thanks by the way for the quick reply.

closed 2017-01-09 by WoodsterDK