convertersystems/opc-ua-client #14: Example for Subscribe to a OPC item via it

jogibear9988, 2017-01-11:


awcullen, 2017-01-11:
Here's an example of data logger. This console program reads a list of nodes from UnifiedAutomation's UACPPServer and logs them to the console. Due to the while loop, you can see the console recovers from errors and reconnects. The first time you run it, you will have to open UA's Server Administrator app and "Trust" the DataLogger's certificate.

https://gist.github.com/awcullen/bb6d4403d07dc43d53a2f74a6b4ae69c

jogibear9988, 2017-01-11:
Its only possible to read directly? Is it not possible to subscribe to a Value, so I get a message everytime it changes?

awcullen, 2017-01-11:
Sure, take a look in sample ConsoleApp: Does this do what you want? 
```
                Console.WriteLine("Step 6 - Create a subscription.");
                var subscriptionRequest = new CreateSubscriptionRequest
                {
                    RequestedPublishingInterval = 1000,
                    RequestedMaxKeepAliveCount = 10,
                    RequestedLifetimeCount = 30,
                    PublishingEnabled = true
                };
                var subscriptionResponse = await session.CreateSubscriptionAsync(subscriptionRequest).ConfigureAwait(false);
                var id = subscriptionResponse.SubscriptionId;

                Console.WriteLine("Step 7 - Add items to the subscription.");
                var itemsToCreate = new MonitoredItemCreateRequest[]
                {
                    new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2258"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }
                };
                var itemsRequest = new CreateMonitoredItemsRequest
                {
                    SubscriptionId = id,
                    ItemsToCreate = itemsToCreate,
                };
                var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);

                Console.WriteLine("Step 8 - Publish the subscription.");
                var publishRequest = new PublishRequest
                {
                    SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
                };
                Console.WriteLine("Press any key to delete the subscription...");
                while (!Console.KeyAvailable)
                {
                    var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

                    // loop thru all the data change notifications
                    var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                    foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            Console.WriteLine($"clientHandle: {min.ClientHandle}; value: {min.Value}");
                        }
                    }

                    publishRequest = new PublishRequest
                    {
                        SubscriptionAcknowledgements = new[] { new SubscriptionAcknowledgement { SequenceNumber = publishResponse.NotificationMessage.SequenceNumber, SubscriptionId = publishResponse.SubscriptionId } }
                    };
                }

                Console.ReadKey(true);

                Console.WriteLine("Step 9 - Delete the subscription.");
                var request = new DeleteSubscriptionsRequest
                {
                    SubscriptionIds = new uint[] { id }
                };
                await session.DeleteSubscriptionsAsync(request).ConfigureAwait(false);
```

jogibear9988, 2017-01-20:
Does that mean, I have to Call session.PublishAsync again every time to get the published results from the opc? is there no event when new data is there?

awcullen, 2017-01-20:
That's correct. The UaTcpSessionChannel does not have a background thread calling PublishAsync for you.
The UaTcpSessionClient does have a background task, but only to support the 'ViewModel with Attributes'.  I see an opportunity for improvement!

jogibear9988, 2017-01-20:
That means also, the opc server does not trigger a tcp message when a monitored node changes, it waits for a publish request??

awcullen, 2017-01-20:
That's true. MonitoredItems just queue up DataValues until you call Publish.

jogibear9988, 2017-01-23:
I've one more question. Is only the first PublishRequest without a Subscription ID? Does this Code also work if i have multiple subscrition id's (because of different refresh times)?

jogibear9988, 2017-01-24:
I get now the following Exception:

Ausnahme ausgelöst: "Workstation.ServiceModel.Ua.ServiceResultException" in mscorlib.dll

Zusätzliche Informationen: The request could not be processed because it specified too many operations.

when I call this:

      _session.CreateMonitoredItemsAsync(itemsRequest).Result;

jogibear9988, 2017-01-24:
One more Question, is there Anywhere a Documentation of the Request Properties available?

jogibear9988, 2017-01-24:
And one more. I've now splitted the _session.CreateMonitoredItemsAsync into 10 Requests, so it works.

But from the first session.PublishAsync(_publishRequest).Result; I get a timeout exception.

awcullen, 2017-01-24:
You can register for free and read the specifications [here](https://opcfoundation.org/developer-tools/specifications-unified-architecture). 
Part 4 - Services has the info about the Request / Response properties.

Actually, PublishRequest does not have a SubscriptionId ever. You never know what subscription you will get! 

Lastly, the PublishRequest is unlike all other Requests in that you should put a long timeout (like 2 minutes) in the RequestHeader.  This is because the Client is expected to send a few PublishRequests in series (fill the pipeline). Then the Server will hold on to them until it has DataChanges or Events to return in a PublishResponse.  When the Client receives the PublishResponse, it responds with a new PublishRequest (with the sequence# of the previous Response as an acknowledgement).

jogibear9988, 2017-01-24:
Does that also mean, a subscription sends the value only when it changes? is it also possible to subscribe, and get the current value once?

and do you know anything about the too many operstions error?

jogibear9988, 2017-01-24:
and i a publish request times out, should i send again the same pub. rq (with the sequence number), or should i send a empty  one?

awcullen, 2017-01-24:
I get the current value first, for every MonitoredItem. I can't find where the Specification says this, or how to find out how many operations is too many.

If a PublishRequest times out, I send a new one, without sending the  Sequence#.

jogibear9988, 2017-01-24:
i try again tomorrow!

jogibear9988, 2017-01-25:
Problem is: I think, without any call to PublishRequest, you do some Publish request internaly in your library. (I've a breakpoint at my PublishAsync call wich is never hit, but in wireshark there are some PublishRequests sent). So the answer to my Monitored Items is not send to my publishrequest async call

awcullen, 2017-01-25:
Apologize for lack of documentation!
Please use UaTcpSessionChannel instead of UaTcpSessionClient.
UaTcpSessionChannel has a simple Request/Response api that matches the UA spec. 

UaTcpSessionClient is the more advanced component that (for now) only works for the ViewModel based subscriptions.

Please see the ConsoleApp for example of using UaTcpSessionChannel.

jogibear9988, 2017-01-26:
seems to work now.

thanks for the information and your great lib!

closed 2017-01-26 by jogibear9988

balamcsd, 2017-06-27:
When monitoring for multiple nodes, how to get the node name, node id, in data changed notification.
I have to get it in below lines of code  :
                  foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            Console.WriteLine($"clientHandle: {min.ClientHandle}; value: {min.Value}");
                        }
                    }