convertersystems/opc-ua-client #18: ViewModelBase

mirkoboggian, 2017-01-20:
Hi, this weekend i will try your lib. I install it using NuGet (i cannot open UaClient project in VS2015) but i cannot found the reference library for the "ViewModelBase" class. Which library do i must use?

awcullen, 2017-01-20:
The reason there is no ViewModelBase in the library is because everyone has a favorite!
Instead of requiring a certain base class, the solution uses Attributes.
Be sure to see the https://github.com/convertersystems/opc-ua-samples

mirkoboggian, 2017-01-20:
Thank you for the quick reply! I'm sorry, i saw your updates a little bit
after i publish the issue.
I'm going to check it another time this weekend. I will feedback you soon.

Thanks again

Tecnoeletronic
dott. Ing. Mirko Boggian

Ufficio:  (+39) 0521 878254
Fax:       (+39) 0521 346219
Cell:      (+39) 338 1456608
E-mail:  mirko@tecnoeletronic.net
sito:      www.tecnoeletronic.net

mirkoboggian, 2017-01-26:
> Ufficio:  (+39) 0521 878254 <+39%200521%20878254>
> Fax:       (+39) 0521 346219 <+39%200521%20346219>
> Cell:      (+39) 338 1456608 <+39%20338%20145%206608>

Hi,

i tryed your library and i can read/write data and subscribe changes well.
I have some questions:

1. in subscriptions: the meaning of PublishingInterval, KeepAliveCount,
 LifetimeCount, PublishingEnable.
2. how to automatic reconnect a UaTcpSessionChannel if errors occurred
3. in subscriptions notifications:
    - OK: if "subscribed data" changes in the OPC server i receive the
right notification with the new value (1 notification with the new value
and timestamp)
    - KO: if "subscribed data" changes calling the
UaTcpSessionChannel.WriteAsync (same UaTcpSessionChannel used for
subscriptions) I receive 2/3 notifications.
     In case of 2 (integer value):
      1) old value with right current Timestamp
      2) new value with the Timestamp = DateTime.MinValue
     In case of 3 (string value):
      1) new value with right current timestamp
      2) old value with the right timestamp
      3) new value with the Timestamp = DateTime.MinValue
Why?

I use  https://github.com/convertersystems/opc-ua-samples source codes to
test it.

Tecnoeletronic
dott. Ing. Mirko Boggian

Ufficio:  (+39) 0521 878254
Fax:       (+39) 0521 346219
Cell:      (+39) 338 1456608
E-mail:  mirko@tecnoeletronic.net
sito:      www.tecnoeletronic.net

2017-01-20 17:44 GMT+01:00 Mirko Boggian <mirko.boggian@gmail.com>:

mirkoboggian, 2017-01-26:
Hi,

here you are a Trace of "Subscription data change" using a WriteAsync
operation:

OpcUA_TestDevice.vshost.exe Information: 0 : Tag:
Conveyors.MachineStatus.CurrentFormat, Value: 114, Timestamp: 01/01/0001
00:00:00, Info: 0x00000000: The operation completed successfully.
OpcUA_TestDevice.vshost.exe Information: 0 : Tag:
Conveyors.MachineStatus.CurrentFormat, Value: 113, Timestamp: 26/01/2017
08:38:45, Info: 0x00000000: The operation completed successfully.
OpcUA_TestDevice.vshost.exe Information: 0 : Tag:
Conveyors.MachineStatus.CurrentFormat, Value: 114, Timestamp: 26/01/2017
08:39:51, Info: 0x00000000: The operation completed successfully.

here you are a Trace of "Subscription data change" changing the value
direct into PLC:

OpcUA_TestDevice.vshost.exe Information: 0 : Tag:
Conveyors.MachineStatus.CurrentFormat, Value: 111, Timestamp: 26/01/2017
08:43:49, Info: 0x00000000: The operation completed successfully.

Before i wrote 2/3 notifications but i was wrong: notifications are always
3 (int/string).

Tecnoeletronic
dott. Ing. Mirko Boggian

Ufficio:  (+39) 0521 878254
Fax:       (+39) 0521 346219
Cell:      (+39) 338 1456608
E-mail:  mirko@tecnoeletronic.net
sito:      www.tecnoeletronic.net

2017-01-26 8:32 GMT+01:00 Mirko Boggian <mirko.boggian@gmail.com>:

awcullen, 2017-01-26:
1. You can register for free and read the specifications [here](https://opcfoundation.org/developer-tools/specifications-unified-architecture). 
Part 4 - Services has the info about all the Request / Response properties.

2. Try out the DataLoggingSolution in the opc-ua-samples repo, It reconnects on error. For instance, if the Server restarts while you are running the logger, the logger reconnects.

3. I tested this behavior with the test servers I have in my office, and did not receive 3 responses. Please tell me more about what Opc Server and PLC you are using?  

I think this a problem of the Opc Server, but you should post the code you are using to call WriteAsync.

Thanks
Andrew

mirkoboggian, 2017-01-26:
Hi,

1) thanks
2) i'm going to check
3) i'm using PLC S71200 and OPC Server Ignition. I'm going to test it using
Kepware.

i will inform you.

Tecnoeletronic
dott. Ing. Mirko Boggian

Ufficio:  (+39) 0521 878254
Fax:       (+39) 0521 346219
Cell:      (+39) 338 1456608
E-mail:  mirko@tecnoeletronic.net
sito:      www.tecnoeletronic.net

awcullen, 2017-01-26:
I recommend trial of:
SIMATIC NET DVD V14   6GK1700-0AA14-0AA0

closed 2017-02-02 by awcullen