convertersystems/opc-ua-client #21: Out Of Memory Exception when compiling (Release) a UWP app referencing workstation.UaClient

TechnoScream, 2017-02-09:
When I try to compile a Release build of a UWP app, the compile fails with the following error.

ILT0005: 'C:\Program Files (x86)\MSBuild\Microsoft\.NetNative\x86\ilc\tools\Gatekeeper.exe @"c:\users\*****\documents\visual studio 2015\Projects\TestUWPOpcUa\TestUWPOpcUa\obj\x86\Release\ilc\intermediate\gkargs.rsp"' returned exception code -532462766	TestUWPOpcUa

This problem does not happen with Debug build. I have attached a blank sample app which reproduces the problem on my side.
[TestUWPOpcUa.zip](https://github.com/convertersystems/opc-ua-client/files/764017/TestUWPOpcUa.zip)

awcullen, 2017-02-09:
When I compile the release build of your project I get the same error. If I change MSBuild Output to 'Detailed', I see that Gatekeeper is having trouble with my dependency 'BouncyCastle'.  I use BouncyCastle for encryption and building certificates, so I'm stuck. Maybe when NetStandard 2.0  arrives I can switch back to Microsoft's library.

On the Build tab of your project's properties, if you switch off the [ ] Compile  with .Net Native tool chain, you will compile successfully for now.

It seems that the Net Native Toolchain will be getting some improvements this year.
http://stackoverflow.com/questions/40828096/error-creating-uwp-store-package-or-release-build-out-of-memory

jogibear9988, 2017-02-09:
siemens also uses bouncycastle in its opcua sample

TechnoScream, 2017-02-10:
Too bad the problem comes from another class. I'll have to use with the work around for now.
Thanks for the quick answer!

MattWhilden, 2017-03-16:
I work on the .NET Native runtime and compiler team. We have a new release (1.6) as of last week. It requires VS 2017 but community edition should be enough. Happy to help if anything's amiss.

Here’s some instructions on how to get to latest of everything: 

- 	 Install VS 2017 RTM and pick the UWP workflow at install
- 	 Open the project
- 	 Open the nuget package manager
- 	 Select the package for Microsoft.NETCore.UniversalWindowApplication (aka the UWP metapackage)
- 	 Update to 5.3.1
- 	 Build and run.

closed 2019-07-28 by awcullen