convertersystems/opc-ua-client #22: opc ua alarms and conditions

abrasat, 2017-02-15:
Does the opc-ua-client support also opc-ua alarms and conditions ?
Does the sample node opcua server support alarms and conditions ?

awcullen, 2017-02-15:
Yes, the opc-ua-client supports alarms and conditions.
When using view-models, you can subscribe to alarms and conditions as you would events.
Here I'm connecting to the UACPPServer and subscribing to two different nodes that are EventNotifiers.  
The property 'Server' is typed as 'BaseEvent', a type with fields for Time, Message, Severity, etc.
The property 'AlarmsWithNodes' is typed as 'AlarmCondition', a sub-class of BaseEvent with additional fields for ActiveState, AcknowlegedState, etc.
 

```
        /// <summary>
        /// Gets the event of AlarmsWithNodes.
        /// </summary>
        [MonitoredItem(nodeId: "ns=2;s=AlarmsWithNodes", attributeId: AttributeIds.EventNotifier)]
        public AlarmCondition AlarmsWithNodes
        {
            get { return this.alarmsWithNodes; }
            private set { this.SetProperty(ref this.alarmsWithNodes, value); }
        }

        private AlarmCondition alarmsWithNodes;

        /// <summary>
        /// Gets the event of Server.
        /// </summary>
        [MonitoredItem(nodeId: "i=2253", attributeId: AttributeIds.EventNotifier)]
        public BaseEvent Server
        {
            get { return this.server; }
            private set { this.SetProperty(ref this.server, value); }
        }

        private BaseEvent server;

```

The sample server does not explore alarms and conditions. But you can find lots of examples at https://github.com/node-opcua

abrasat, 2017-02-28:
Thank you for the sample.
Do you eventually have a opc-ua client sample about how to browse/list all nodes having set the EventNotifier attribute to SubscribeToEvents, from a opc-ua server ?

awcullen, 2017-02-28:
The only way to do what you ask is to Browse for nodes of type Object, read the EventNotifier attribute,  and then repeat for the entire server Namespace. Often I find programs that Browse the entire Namespace to be slow.
More importantly, the 'Server' object reports all the Events. Any other EventNotifier you find in the namespace is just reporting a subset of the Events.

awcullen, 2017-02-28:
Well, I can see now that the sample above was not clear about one thing. You are free to choose any subtype of BaseEvent as the property type. Any event fields that are not reported by the original event will be set to Null.

In the following example I have coded that all events that are reported by the 'Server' node will be received as type 'AlarmCondition'. If the original event does not have a field for 'AcknowlegedState', for example, it will be set to Null.
 
```
        /// <summary>
        /// Gets the event of Server.
        /// </summary>
        [MonitoredItem(nodeId: "i=2253", attributeId: AttributeIds.EventNotifier)]
        public AlarmCondition Server
        {
            get { return this.server; }
            private set { this.SetProperty(ref this.server, value); }
        }

        private AlarmCondition server;
```

abrasat, 2017-02-28:
Thanks again.
I think that it would be practical to be able to read the Server_NamespaceArray at the client start, to have later the possibility to filter events (or other objects) by their node namespace, for instance when browsing.
Actually the server namespace index is not guaranteed to be immutable, the safest way for the client would be to get the index dynamically by reading the Server_NamespaceArray property (of course the namespace URIs handled by the opc-ua server have to be well-known and are not allowed to change; just the index in the server namespace table might change).
I could not figure out how to read the Server_NamespaceArray property, it would be helpful if you have a sample for it.

awcullen, 2017-02-28:
If you use the UaTcpSessionChannel the Server_NamespaceArray is already available at property NamespaceUris.

```
            // fetch namespace array, etc.
            var readValueIds = new ReadValueId[]
            {
                new ReadValueId
                {
                    NodeId = NodeId.Parse(VariableIds.Server_NamespaceArray),
                    AttributeId = AttributeIds.Value
                },
                new ReadValueId
                {
                    NodeId = NodeId.Parse(VariableIds.Server_ServerArray),
                    AttributeId = AttributeIds.Value
                }
            };
            var readRequest = new ReadRequest
            {
                NodesToRead = readValueIds
            };

            var readResponse = await this.ReadAsync(readRequest).ConfigureAwait(false);
            if (readResponse.Results.Length == 2)
            {
                if (StatusCode.IsGood(readResponse.Results[0].StatusCode))
                {
                    this.NamespaceUris.Clear();
                    this.NamespaceUris.AddRange(readResponse.Results[0].GetValueOrDefault<string[]>());
                }

                if (StatusCode.IsGood(readResponse.Results[1].StatusCode))
                {
                    this.ServerUris.Clear();
                    this.ServerUris.AddRange(readResponse.Results[1].GetValueOrDefault<string[]>());
                }
            }
```

abrasat, 2017-03-01:
Thank you, it worked.
I think that it would be a good and useful option in opc-ua client to extend the NodeId (and especially the ViewModel NodeId attribute) with the possibility to work additionally with namespaceUri instead of only with namespace index (I mean the opc-ua client can get the actual namespace index "behind the scenes", based on the Namespace Uri).
From the opc-ua client application point of view it would be better in my opinion to store together with the node identifiers the namespaceUris and not the namespace Indexes.

awcullen, 2017-03-01:
Maybe [ExpandedNodeId](https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/ExpandedNodeId.cs) will meet your needs? 
Maybe I should have used it in the MonitoredItemAttribute.
When using UaTcpSessionChannel, I am following the [specifications](https://opcfoundation.org/developer-tools/specifications-unified-architecture), which uses NodeId mostly everywhere.

abrasat, 2017-03-01:
Yes, I see now that ExpandedNodeId works with namespace uri.
You could eventually add an additional MonitoredItemExpAttribute based on ExpandedNodeId, if you do not want to break compatibility.

awcullen, 2017-03-01:
You will just have to inspect the code for [EventMonitoredItem](https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/MonitoredItemBase.cs#L259) and [BaseEvent](https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/BaseEvent.cs) to see the inner workings
.
BaseEvent has a field for EventType which is the NodeId of the event type.

abrasat, 2017-03-02:
Thanks, I will take a look. 
The subscription via "MonitoredItem" attribute is working fine.
But at the first sight is not obvious how to read/publish event notifier nodes in a similar way with how the server properties or the application values are read with ReadAsync() or published with PublishAsync().

abrasat, 2017-03-03:
After inspecting the code, I could read event notifiers also without using the MonitoredItem attributes.
Are you planning an extension of the MonitoredItem attribute to work also with ExpandedNodeId ?

awcullen, 2017-03-03:
Yes, I'm thinking it would good for MonitoredItem attribute accept both NodeId and ExpandedNodeID

abrasat, 2017-03-04:
Ok, great news.
Is there anywhere in code a sample about how to acknowledege an alarm ? 
Does the opc-ua client support also event history ?

awcullen, 2017-03-11:
Code to acknowledge an alarm
The problem now is finding the conditionId. If the AlarmCondition is browsable in the namespace, then you can use the NodeId of the instance. But alarms are not required to be nodes in the namespace. In the next version, I will add a ConditionId property to the Condition class. Then you will be able to acknowledge any alarm.
```
        /// <summary>
        /// Invokes the method AlarmAcknowledge.
        /// </summary>
        public async Task AlarmAcknowledge(NodeId conditionId, byte[] eventId, LocalizedText comment)
        {
            var response = await this.session.CallAsync(new CallRequest
            {
                MethodsToCall = new[]
                {
                    new CallMethodRequest
                    {
                        ObjectId = conditionId,
                        MethodId = NodeId.Parse(MethodIds.AcknowledgeableConditionType_Acknowledge),
                        InputArguments = new[] { new Variant(eventId), new Variant(comment) }
                    }
                }
            });
            var result = response.Results[0];
            if (StatusCode.IsBad(result.StatusCode))
            {
                throw new ServiceResultException(new ServiceResult(result.StatusCode));
            }
        }
```

abrasat, 2017-04-24:
Would be possible to define in the view-model a List/ObservableCollection of BaseEvent and AlarmCondition objects (instead of single BaseEvent and AlarmCondition objects), and bind it for instance to a grid view where all actual alarms and events are visualized ?

awcullen, 2017-04-24:
In the view-model, define a public ObservableCollection<AlarmCondition> property to hold the AlarmConditions and BaseEvents. Add a private AlarmCondition property with a setter that adds it's value to the collection. Add a MonitoredItem attribute to this write-only property.

```
        /// <summary>
        /// Gets the current events of Robot1.
        /// </summary>
        public ObservableCollection<AlarmCondition> Robot1Events { get; } = new ObservableCollection<AlarmCondition>();

        [MonitoredItem(nodeId: "ns=2;s=Robot1", attributeId: AttributeIds.EventNotifier)]
        private AlarmCondition Robot1EventStream
        {
            set
            {
                    this.Robot1Events.Add(value);
            }
        }
```

What do you think should be the condition to remove the alarm or event from the collection? Maximum count?, time?

abrasat, 2017-04-25:
Thank you for the code sample. Eventually a converter will be needed, to adapt the different AlarmConditions/BaseEvent fields to the information displayed in the columns of the grid-view.
Regarding your question, I would say that dependently on the custom requirements both options should be possible: either max. number of alarms/events, or a time interval (like the last 24 hours).
Are the OPC-UA servers not supposed to hold alarms/events as history data ? What happens if in a special case the user wants to see the alarms/events from last month ?
Would be possible to access these "old" events by the opc-ua client ? (actually I opened already a issue with this question)

awcullen, 2017-04-25:
You have two good questions here. 
About the converter/ view-model: I wanted to highlight in the example above, that each item of the ObservableCollection<AlarmCondition> will be an instance of AlarmCondition. There may be some events received that are base events, like SystemEvents. In this case the OPC server sends all fields of AlarmCondition, but sets the missing fields to null. To identify whether you received an AlarmCondition or a SystemEvent, you can see the EventType property. The result is that you should really use a DataTemplate and then add a DataTrigger that is bound to the EventType property that shows/hides the Acknowledge button, for instance.

About event history, I have had no time to learn about it. The UaTcpSessionChannel supports the HistoryReadAsync method. I encourage you to try it, and post what you find out.

awcullen, 2017-04-26:
Closing issue for now.

closed 2017-04-26 by awcullen

abrasat, 2017-07-11:
Could you please post a code snippet, about how exactly can be used the EventType property to identify if the received item is a AlarmCondition or a BaseEvent ? I cannot figure out how to peform the check.

awcullen, 2017-07-11:
The EventType property is a NodeId.  You can compare the value to known EventType ids. You can use the ObjectTypeIds enum to generate the NodeIds your looking for like so:
NodeId baseEventTypeId = NodeId.Parse(ObjectTypeIds.BaseEventType);
NodeId alarmConditionEventTypeId = NodeId.Parse(ObjectTypeIds.AlarmConditionType);

abrasat, 2017-07-12:
OK, thanks.
Actually there are much more types to check, like the AlarmConditionType "parent" types ConditionType, AcknowledgeableConditionType, or "child" types "DiscreteAlarmType", "LimitAlarmType".