convertersystems/opc-ua-client #23: opc-ua-client NetStandard version test

abrasat, 2017-02-17:
Could you please add also test methods for the opc-ua-client NetStandard version ?
I have some problems  with an adapted Xamarin Droid sample (which uses the NetStandard opc-ua-client version) to subscribe via attributes to opc-ua nodes from out custom opc-ua-server (the "session.Subscribe()" call crashes).
I added a test method to subscribe via attributes to opc-ua nodes from out custom opc-ua-server by using the Net45 opc-ua-client and this is working fine.

Is there any possibility to debug the opc-ua-client NetStandard version (nuget package with debug information ?)

awcullen, 2017-02-17:
I would like to learn more about the problem with Subscribe. Could you provide the Exception info and the output from the Debug window.

To debug opc-ua-client with your project:
First, right-click your Solution and use "Manage Nuget Packages" to remove the Workstation.UaClient.
Second, go to https://github.com/convertersystems/opc-ua-client and either download (and unzip) -or- clone the project.
Third, right-click your Solution and use "Add... Existing Project...". Browse to the project 'Workstation.UaClient.shproj' and click Ok. Repeat for project 'Workstation.UaClient.NetStandard.csproj'
Fourth,  right-click your project's References folder and Add a reference to Project...Solution... Workstation.UaClient.NetStandard.

abrasat, 2017-02-17:
Here is what I see in the Debug window (only the opc-ua-client log messages). I guess there is some error in my sample droid application.

`[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://xx.xx.xx.xx:yyyyy'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentityToken: 'Workstation.ServiceModel.Ua.AnonymousIdentity'.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending OpenSecureChannelRequest Handle: 2
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received OpenSecureChannelResponse Handle: 2 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending CreateSessionRequest Handle: 3
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Installed new security token 1.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received CreateSessionResponse Handle: 3 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ActivateSessionResponse Handle: 4 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending ReadRequest Handle: 5
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ReadResponse Handle: 5 Result: 0x00000000
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opened.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 6
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 7
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 8
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 6 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 7 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 8 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 9
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 9 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 10
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Sending PublishRequest Handle: 11
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 10 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.
[0:] Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Received ServiceFault Handle: 11 Result: 0x80790000
[0:] Workstation.ServiceModel.Ua.UaTcpSessionClient: Error: Error publishing subscription. There is no subscription available for this session.`

abrasat, 2017-02-17:
I fixed the problem, was an error in my application.
Thank you for your support !

closed 2017-02-17 by abrasat