convertersystems/opc-ua-client #25: Support for arrays

abrasat, 2017-03-09:
Is it possible to have have a MonitoredItem attribute with a nodeId corresponding to an array (of float values for instance) ? How does the attribute syntax look like ?
How can be written each array element individually by using WriteValues with the WriteAsync() method ?

awcullen, 2017-03-10:
Yes, the attribute syntax is:
```
/// <summary>
/// Gets or sets the value of ArraysFloat.
/// </summary>
[MonitoredItem(nodeId: "ns=2;s=Demo.Static.Arrays.Float")]
public Single[] ArraysFloat
{
    get { return this.arraysFloat; }
    set { this.SetProperty(ref this.arraysFloat, value); }
}

private Single[] arraysFloat;
```

You can write an individual element by setting the property WriteValue property "IndexRange"
IndexRange is a string. It supports writing a single element, example "1", which means the second element of the array,
or a range of elements, example "1:2", which means return the second and third element.
The things to remember is that the array indexes start at zero,
and the value written must still be an array, even if you are writing a single element.
```
                var req = new WriteRequest
                {
                    NodesToWrite = new WriteValue[]
                    {
                        new WriteValue
                        {
                            NodeId = NodeId.Parse("ns=2;s=Demo.Static.Arrays.Float"),
                            IndexRange = "1",  // array indexes start from zero
                            AttributeId = AttributeIds.Value,
                            Value = new DataValue(new float[] { 1.0f }) // must still be an array 
                        }
                    }
                };
                var res = await this.session.WriteAsync(req);

```

abrasat, 2017-03-11:
Thank you for the clarification.
Is the support of indexRange optional for opc-ua servers? If yes, is there any possibility to check if the server supports it or not ?
Is it possible to use the indexRange also in the attribute for view model, in view to access only one array element?

awcullen, 2017-03-23:
I don't know if indexRange is optional.

You can add a 'indexRange' property to MonitoredItemAttribute. But please note that adding an indexRange of one element does not change the data type. It still returns an array. An array with just one element.

closed 2017-03-23 by awcullen