convertersystems/opc-ua-client #26: Issue while installing Workstation.UaClient nuget in UWP application

tejasklodha, 2017-03-11:
Dear Sir,

I tried to install nuget Workstation.UaClient in My UWP application but give error that  "Package restore failed. Rolling back package changes for 'MyProjectName."

I am able successfully install nuget in Xamarin Form App, WPF and console application...
Only face issue in UWP app also tried to make new project with only UWP application but still facing issue..
Are you done Any new changes regarding UWP application?

Thanks,

awcullen, 2017-03-11:
The [samples](https://github.com/convertersystems/opc-ua-samples) include a UWP project 'RobotApp'.
I uninstalled and reinstalled the UaClient package with no problems.

tejasklodha, 2017-03-14:
I found out the solution after struggling in this weekend....

Resolution is very simple that when I understood the issue.
Issue is when new UWP project created then automatically use UniversalWindowsPlatform 5.1.0 nuget.
Solution is I upgraded UniversalWindowsPlatform- 5.1.0 nuget to UniversalWindowsPlatform 5.2.2
And Problem Resolved.

Tejas Lodha,

closed 2017-03-14 by tejasklodha