convertersystems/opc-ua-client #28: Browse does not find all Items on my OPC

jogibear9988, 2017-04-04:
If I use your Visual Studio Addin, or my COde, I can not find all Datablocks on my PLC. If I use the Client from https://www.unified-automation.com/downloads/opc-ua-clients/file/download/details/uaexpert-v140.html it works. (and is much faster in loading the information)

awcullen, 2017-04-04:
Please tell me more about the missing blocks. Is it always the same blocks? Would it be possible to send a screenshot?

About performance, UaExpert browses the nodes first. Then, if you select the node, UaExpert reads the attributes. UaBrowser both browses and reads the attributes (datatype, arrayrank, etc) at once. I prefer to see the datatype when I'm browsing. But I have not used a 1500. Is the UaBrowser performance too slow on a 1500?

awcullen, 2017-04-04:
I uploaded a new version 1.5.9 of UaBrowser to the Marketplace. (Maybe Visual Studio will notify you?)
This contains the bugfix for the DataLengthException that you reported (and took me a while to fix).
Maybe this is related.

jogibear9988, 2017-04-04:
The new Version works....

awcullen, 2017-04-04:
Fantastic!

jogibear9988, 2017-04-04:
When I use a newer Version of your Assembly as 1.5.7 I got this Exception:

System.IO.FileLoadException: Die Datei oder Assembly "Microsoft.Extensions.Logging.Abstractions, Version=1.1.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60" oder eine Abhängigkeit davon wurde nicht gefunden. Die gefundene Manifestdefinition der Assembly stimmt nicht mit dem Assemblyverweis überein. (Ausnahme von HRESULT: 0x80131040)
Dateiname: "Microsoft.Extensions.Logging.Abstractions, Version=1.1.0.0, Culture=neutral, PublicKeyToken=adb9793829ddae60" ---> System.IO.FileLoadException: Die Datei oder Assembly "Microsoft.Extensions.Logging.Abstractions, Version=1.1.1.0, Culture=neutral, PublicKeyToken=adb9793829ddae60" oder eine Abhängigkeit davon wurde nicht gefunden. Die gefundene Manifestdefinition der Assembly stimmt nicht mit dem Assemblyverweis überein. (Ausnahme von HRESULT: 0x80131040)
Dateiname: "Microsoft.Extensions.Logging.Abstractions, Version=1.1.1.0, Culture=neutral, PublicKeyToken=adb9793829ddae60"

awcullen, 2017-04-04:
Yes, 1.5.9 requests "Microsoft.Extensions.Logging.Abstractions, Version=1.1.1.0"
Will that be a problem?

jogibear9988, 2017-04-04:
i've solved it with a app.config entry!

jogibear9988, 2017-04-04:
one more problem, i've a timeout for my request! can i increase it?

awcullen, 2017-04-04:
If you are talking about UaBrowser, then unfortunately I don't expose the timeouts now. The default is 15 seconds for each request. Too small? Would 30 seconds be enough?

If you are using the UaTcpSessionChannel in code, then modify the constructor
```
                                this.session = new UaTcpSessionChannel(
                                this.localDescription,
                                this.CertificateStore,
                                this.userIdentity,
                                selectedEndpoint, 
                                timeoutHint: 30000);
```

jogibear9988, 2017-04-05:
Is it normal that a subscribe to 1000 Tags takes about 30 seconds?

awcullen, 2017-04-05:
I installed UaCPPServer from http://www.unified-automation.com/ on a separate PC.
I modified the ConsoleApp from my opc-ua-samples repo, to create a subscription 
with 1000 monitored items. 
The stopwatch runs 22 - 24 milliseconds.

Security set to Basic256 - Sign and Encrypt

```
var watch = new Stopwatch();
var itemsToCreate = new MonitoredItemCreateRequest[1000];
for (int i = 0; i < 1000; i++)
{
    itemsToCreate[i] =
        new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse($"ns=2;s=Demo.Massfolder_Dynamic.Variable{string.Format("0000", i)}"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = (uint)i, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } };
}

Console.WriteLine("Step 6 - Create a subscription.");
watch.Restart();

var subscriptionRequest = new CreateSubscriptionRequest
{
    RequestedPublishingInterval = 1000,
    RequestedMaxKeepAliveCount = 10,
    RequestedLifetimeCount = 30,
    PublishingEnabled = true
};
var subscriptionResponse = await session.CreateSubscriptionAsync(subscriptionRequest).ConfigureAwait(false);
var id = subscriptionResponse.SubscriptionId;

Console.WriteLine("Step 7 - Add items to the subscription.");
var itemsRequest = new CreateMonitoredItemsRequest
{
    SubscriptionId = id,
    ItemsToCreate = itemsToCreate,
};
var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);
watch.Stop();
Console.WriteLine($"ElapsedMilliseconds: {watch.ElapsedMilliseconds}");
```

jogibear9988, 2017-04-13:
i've talked to siemens! their opc has limited performance

rebj1011, 2018-08-07:
Hey Andrew,
i've the same porblem as jogi. If i start browsing my Siemens OpcUA server, i only get a few DBs. 
I'm using Visual Studio 17 including the .Net Framework 4.6.1 and your Workstation.UaBrowser fom the Marketplace. 
Could you pls help me?

awcullen, 2018-08-07:
You can also turn on logging to write any exceptions to a file:

1. Start notepad.exe as Administrator (by right click Run as Administrator).

2. Select File...Open.  Change Filter to "All files (* .*). Change directory to  "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE".  Select "devenv.exe.config".  (If that location is not correct, you can search for 'devenv.exe' in the Taskbar, then right click and choose Open file location..."). 

3. Edit the xml to end with:
```
    <system.diagnostics>
      <trace autoflush="false" indentsize="4">
        <listeners>
          <add name="myListener" type="System.Diagnostics.TextWriterTraceListener" initializeData="C:\Users\Public\Documents\devenv.log" />
          <remove name="Default" />
        </listeners>
      </trace>
    </system.diagnostics>

  </configuration>
```
4. Save the file. Restart Visual Studio.

5. Use UABrowser and cause the problem. 

6. Use Explorer to see that the "devenv.log" file is created in "C:\Users\Public\Documents" directory.

rebj1011, 2018-08-08:
Ua Expert Resulat the list of the DBs is even longer that this snippet:
[image]

Result of the "Workstation ConsolApp" Code
``Step 1 - Describe this app.
Step 2 - Create a certificate store.
Step 3 - Create a session with your server.
Server is requesting UserName identity...
Enter user name:
Enter password: 
  Opened channel with endpoint '                                             '.
  SecurityPolicyUri: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
  SecurityMode: 'SignAndEncrypt'.
  UserIdentity: 'Workstation.ServiceModel.Ua.UserNameIdentity'.
Press any key to continue...
Step 4 - Browse the server namespace.
+ Root
  + C: 2:/Plc/C, Variable
  + Clock: 2:/Plc/Clock, Variable
  + I: 2:/Plc/I, Variable
  + IB: 2:/Plc/IB, Variable
  + ID: 2:/Plc/ID, Variable
  + IW: 2:/Plc/IW, Variable
  + M: 2:/Plc/M, Variable
  + MB: 2:/Plc/MB, Variable
  + MD: 2:/Plc/MD, Variable
  + MW: 2:/Plc/MW, Variable
  + Q: 2:/Plc/Q, Variable
  + QB: 2:/Plc/QB, Variable
  + QD: 2:/Plc/QD, Variable
  + QW: 2:/Plc/QW, Variable
  + T: 2:/Plc/T, Variable
  + connect_state: 2:/Plc/connect_state, Variable
  + DB1: 2:/Plc/DB1, Object
    + DB1.DBB: 2:/Plc/DB1.DBB, Variable
    + DB1.DBW: 2:/Plc/DB1.DBW, Variable
    + DB1.DBD: 2:/Plc/DB1.DBD, Variable
    + DB1.DBX: 2:/Plc/DB1.DBX, Variable
  + DB2: 2:/Plc/DB2, Object
    + DB2.DBB: 2:/Plc/DB2.DBB, Variable
    + DB2.DBW: 2:/Plc/DB2.DBW, Variable
    + DB2.DBD: 2:/Plc/DB2.DBD, Variable
    + DB2.DBX: 2:/Plc/DB2.DBX, Variable
  + DB3: 2:/Plc/DB3, Object
    + DB3.DBB: 2:/Plc/DB3.DBB, Variable
    + DB3.DBW: 2:/Plc/DB3.DBW, Variable
    + DB3.DBD: 2:/Plc/DB3.DBD, Variable
    + DB3.DBX: 2:/Plc/DB3.DBX, Variable
  + DB4: 2:/Plc/DB4, Object
    + DB4.DBB: 2:/Plc/DB4.DBB, Variable
    + DB4.DBW: 2:/Plc/DB4.DBW, Variable
    + DB4.DBD: 2:/Plc/DB4.DBD, Variable
    + DB4.DBX: 2:/Plc/DB4.DBX, Variable
  + DB5: 2:/Plc/DB5, Object
    + DB5.DBB: 2:/Plc/DB5.DBB, Variable
    + DB5.DBW: 2:/Plc/DB5.DBW, Variable
    + DB5.DBD: 2:/Plc/DB5.DBD, Variable
    + DB5.DBX: 2:/Plc/DB5.DBX, Variable
  + DB7: 2:/Plc/DB7, Object
    + DB7.DBB: 2:/Plc/DB7.DBB, Variable
    + DB7.DBW: 2:/Plc/DB7.DBW, Variable
    + DB7.DBD: 2:/Plc/DB7.DBD, Variable
    + DB7.DBX: 2:/Plc/DB7.DBX, Variable
  + DB8: 2:/Plc/DB8, Object
    + DB8.DBB: 2:/Plc/DB8.DBB, Variable
    + DB8.DBW: 2:/Plc/DB8.DBW, Variable
    + DB8.DBD: 2:/Plc/DB8.DBD, Variable
    + DB8.DBX: 2:/Plc/DB8.DBX, Variable
  + DB10: 2:/Plc/DB10, Object
    + DB10.DBB: 2:/Plc/DB10.DBB, Variable
[... 115 lines cut ...]
    + DB86.DBW: 2:/Plc/DB86.DBW, Variable
    + DB86.DBD: 2:/Plc/DB86.DBD, Variable
    + DB86.DBX: 2:/Plc/DB86.DBX, Variable
  + DB87: 2:/Plc/DB87, Object
    + DB87.DBB: 2:/Plc/DB87.DBB, Variable
    + DB87.DBW: 2:/Plc/DB87.DBW, Variable
    + DB87.DBD: 2:/Plc/DB87.DBD, Variable
    + DB87.DBX: 2:/Plc/DB87.DBX, Variable
  + DB88: 2:/Plc/DB88, Object
    + DB88.DBB: 2:/Plc/DB88.DBB, Variable
    + DB88.DBW: 2:/Plc/DB88.DBW, Variable
    + DB88.DBD: 2:/Plc/DB88.DBD, Variable
    + DB88.DBX: 2:/Plc/DB88.DBX, Variable
  + DB96: 2:/Plc/DB96, Object
    + DB96.DBB: 2:/Plc/DB96.DBB, Variable
    + DB96.DBW: 2:/Plc/DB96.DBW, Variable
    + DB96.DBD: 2:/Plc/DB96.DBD, Variable
    + DB96.DBX: 2:/Plc/DB96.DBX, Variable
Press any key to continue...
Step 5 - Create a subscription.
Step 6 - Add items to the subscription.
Step 7 - Subscribe to PublishResponse stream.
Press any key to delete the subscription...
Step 8 - Delete the subscription.
Press any key to close the session...

rebj1011, 2018-08-08:
Sadly Visual Studio doesn't create a devenv.log file.

rebj1011, 2018-08-08:
Sadly Visual Studio doesn't create a devenv.log file. There is no error and the ConsoleApp runs till the end.

awcullen, 2018-08-08:
I think the issue is that OpcUa servers can set a limit on the number of nodes that are returned in a BrowseResponse.  I believe the embedded S7 server might do that.  My guess that the number is fifty :)

The way you ask for the next set of nodes is to call BrowseNext with a ContinuationPoint from the previous response.  Below I coded an example of doing that to three levels. 

```
Console.WriteLine("Step 4 - Browse the server namespace.");
Console.WriteLine("+ Root");
var rds1 = new List<ReferenceDescription>();
var browseRequest = new BrowseRequest { NodesToBrowse = new[] { new BrowseDescription { NodeId = NodeId.Parse(ObjectIds.RootFolder), ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), ResultMask = (uint)BrowseResultMask.TargetInfo, NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, BrowseDirection = BrowseDirection.Forward, IncludeSubtypes = true } } };
var browseResponse = await channel.BrowseAsync(browseRequest);
rds1.AddRange(browseResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
var cps = browseResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null).ToArray();
while (cps.Length > 0)
{
    var browseNextRequest = new BrowseNextRequest { ContinuationPoints = cps, ReleaseContinuationPoints = false };
    var browseNextResponse = await channel.BrowseNextAsync(browseNextRequest);
    rds1.AddRange(browseNextResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
}
foreach (var rd1 in rds1)
{
    Console.WriteLine("  + {0}: {1}, {2}", rd1.DisplayName, rd1.BrowseName, rd1.NodeClass);
    var rds2 = new List<ReferenceDescription>();
    browseRequest = new BrowseRequest { NodesToBrowse = new[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd1.NodeId, channel.NamespaceUris), ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), ResultMask = (uint)BrowseResultMask.TargetInfo, NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, BrowseDirection = BrowseDirection.Forward, IncludeSubtypes = true } } };
    browseResponse = await channel.BrowseAsync(browseRequest);
    rds2.AddRange(browseResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
    cps = browseResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null).ToArray();
    while (cps.Length > 0)
    {
        var browseNextRequest = new BrowseNextRequest { ContinuationPoints = cps, ReleaseContinuationPoints = false };
        var browseNextResponse = await channel.BrowseNextAsync(browseNextRequest);
        rds2.AddRange(browseNextResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
    }
    foreach (var rd2 in rds2)
    {
        Console.WriteLine("    + {0}: {1}, {2}", rd2.DisplayName, rd2.BrowseName, rd2.NodeClass);
        var rds3 = new List<ReferenceDescription>();
        browseRequest = new BrowseRequest { NodesToBrowse = new[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd2.NodeId, channel.NamespaceUris), ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), ResultMask = (uint)BrowseResultMask.TargetInfo, NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, BrowseDirection = BrowseDirection.Forward, IncludeSubtypes = true } } };
        browseResponse = await channel.BrowseAsync(browseRequest);
        rds3.AddRange(browseResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
        cps = browseResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null).ToArray();
        while (cps.Length > 0)
        {
            var browseNextRequest = new BrowseNextRequest { ContinuationPoints = cps, ReleaseContinuationPoints = false };
            var browseNextResponse = await channel.BrowseNextAsync(browseNextRequest);
            rds3.AddRange(browseNextResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
        }
        foreach (var rd3 in rds3)
        {
            Console.WriteLine("      + {0}: {1}, {2}", rd3.DisplayName, rd3.BrowseName, rd3.NodeClass);
        }
    }
}
```

rebj1011, 2018-08-09:
Thanks mate now everything works as expected. But three are not enough ;).
You can close this topic.

closed 2018-08-09 by awcullen