convertersystems/opc-ua-client #29: Self created certificate is valid from future timestamp (2h future)

BlackyPanther, 2017-04-04:
__Environment__
- .NET 4.6 Framework
- Visual Studio 2015 Enterprise
- Location: Germany (summer time)
- KepWare KEPServerEX 5 (v5.21.112.0)

__Problem__
Starting with no certificates at the server side and connecting to the server causes the UaClient (DirectoryStore) to create a new self-signed certificate.    
This certificate is sent to the server an can be set to "trusted".    
Until this point everything works just fine. (Thank you by the way for this NuGet package!).

As I tried to connect again to browse the device, I still had a verification error.    

__Approach__
I now figured out, that there is a time collision for the certificate (screenshot below).    
I think it has to do with [Line 159 in file DirectoryStore.cs](https://github.com/convertersystems/opc-ua-client/blob/master/UaClient/ServiceModel/Ua/DirectoryStore.cs#L159).    
I think the generator uses UTC timestamps and there a local timestamp is set. The problem may be fixed by changing the line to `cg.SetNotBefore(DateTime.UtcNow);`.

[image]

awcullen, 2017-04-04:
Thanks for finding and solving the error! 🥇🥇🥇
I am sure there are more...

BlackyPanther, 2017-04-04:
Now it works like charm 👍

closed 2017-04-04 by BlackyPanther