convertersystems/opc-ua-client #30: Publish Request -> gets only One answer every two seconds

jogibear9988, 2017-04-05:
I send a Subscription with about 2700 Items to my OPC.
Later I send a Publish Request. this is answered after 2 seconds. after that I send the next publish request, wich takes again 2 seconds to be answered....

Can I speed up this?

jogibear9988, 2017-04-05:
Also sending mor publish requests at once does not help

awcullen, 2017-04-05:
First, what are your values for:
RequestedPublishingInterval 
RequestedMaxKeepAliveCount 
revisedPublishingInterval 
revisedMaxKeepAliveCount 

Second, are you getting a PublishResponse with a SubscriptionId that matches the created subsriptionId?

Third, are you getting DataChangeNotifications in the PublishResponse, or is the collection empty( indicates a Keep-Alive response)

Andrew

```
    var req = new CreateSubscriptionRequest
    {
        RequestedPublishingInterval = 1000, // 1 second
        RequestedMaxKeepAliveCount = 20, // 20 * publishingInterval
        PublishingEnabled = true
    };
    var res = await channel.CreateSubscriptionAsync(req);
    var subscriptionId = res.SubscriptionId;
    var revisedPublishingInterval = res.RevisedPublishingInterval;
    var revisedMaxKeepAliveCount = res.RevisedMaxKeepAliveCount;
```

jogibear9988, 2017-04-05:
1.) 
RequestedPublishingInterval = 500 but I also tried 10 and  0
RequestedMaxKeepAliveCount = 10
revisedPublishingInterval = 1000
revisedMaxKeepAliveCount = 10

2.) Yes

3.) No, no empty Responses

awcullen, 2017-04-05:
Well, your server revised the PublishingInterval to 1000 ms. So apparently that's the fastest it will go.

Could you try changing SamplingInterval when you create the items.
```
new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse($"ns=2;s=Demo.Massfolder_Dynamic.Variable{string.Format("0000", i)}"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = (uint)i, SamplingInterval = 1000, QueueSize = 1, DiscardOldest = true } };
```

jogibear9988, 2017-04-05:
I've now change dit on the server to 50ms (Siemens PLC). It responds with 500 (don't know why).
But if I do so, a answer now only contains half of the Items (14 instead of 28), so it's not really faster...

awcullen, 2017-04-05:
It's a very, very Mad World

jogibear9988, 2017-04-05:
;-)

jogibear9988, 2017-04-05:
What is the PublishingEnabled in the CreateSubscriptionRequest?

awcullen, 2017-04-05:
You can start the subscription in a mode where it samples its data, stores it in queues, but doesn't publish.

jogibear9988, 2017-04-05:
So I think it's no issue with your Bib, seems the Plc is too slow... 😢

closed 2017-04-06 by awcullen

jogibear9988, 2017-04-06:
thanks for your help & your lib