convertersystems/opc-ua-client #32: Identification of DataChangedNotification elements (e.g. by NodeID)

BlackyPanther, 2017-04-18:
I use most of your example code to receive data from a OPC-UA server.

Subscribing:
```
List<MonitoredItemCreateRequest> items = new List<MonitoredItemCreateRequest>();
foreach (var node in nodeList)
{
	items.Add(new MonitoredItemCreateRequest
	{
		ItemToMonitor = new ReadValueId
		{
			NodeId = NodeId.Parse(node.NodeId),
			AttributeId = AttributeIds.Value
		},
		MonitoringMode = MonitoringMode.Reporting,
		RequestedParameters = new MonitoringParameters
		{
			SamplingInterval = -1,
			QueueSize = 0,
			DiscardOldest = true,
			ClientHandle = clientHandle
		}
	});
}

var itemsRequest = new CreateMonitoredItemsRequest
{
	SubscriptionId = id,
	ItemsToCreate = items.ToArray()
};
var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);
```

Request the data:
```
publishRequest.SubscriptionAcknowledgements[0] = new SubscriptionAcknowledgement
{
	SubscriptionId = subscriptionId
};
var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);
subscriptionSequences[subscriptionId] = publishResponse.NotificationMessage.SequenceNumber;
var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
var items = dcns.First().MonitoredItems;
```

How is it possible to identify the received data (which node id was it from) ?
As I always receive only a subset of the subscribed nodes (as it should be... due only some data changed).

Thank you for your help!

awcullen, 2017-04-18:
Hi,
There are many confusing details of the OPC-UA specification.  The first is that PublishResponse returns data for any of the subscriptions you have created. You have no control over which subscription is Published next if you have more than one subscription created and enabled. 

The second detail is that each data change or event returned in the PublishResponse is identified only by the SubscriptionId and ClientHandle, a uint number that you generate, that uniquely identifies the MonitoredItem. 

In the example below, there is a while loop that demonstrates how to fill the PublishRequest (with data from previously received responses that you wish to acknowledge) and the how to read data from a PublishResponse.

```
Console.WriteLine("Step 6 - Create a subscription.");
var subscriptionRequest = new CreateSubscriptionRequest
{
    RequestedPublishingInterval = 1000,
    RequestedMaxKeepAliveCount = 10,
    RequestedLifetimeCount = 30,
    PublishingEnabled = true
};
var subscriptionResponse = await session.CreateSubscriptionAsync(subscriptionRequest).ConfigureAwait(false);
var id = subscriptionResponse.SubscriptionId;

Console.WriteLine("Step 7 - Add items to the subscription.");
var itemsToCreate = new MonitoredItemCreateRequest[]
{
    new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2258"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }
};
var itemsRequest = new CreateMonitoredItemsRequest
{
    SubscriptionId = id,
    ItemsToCreate = itemsToCreate,
};
var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);

Console.WriteLine("Step 8 - Publish the subscription.");
var publishRequest = new PublishRequest
{
    SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
};
Console.WriteLine("Press any key to delete the subscription...");
while (!Console.KeyAvailable)
{
    var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

    // loop thru all the data change notifications
    var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
    foreach (var dcn in dcns)
    {
        foreach (var min in dcn.MonitoredItems)
        {
            Console.WriteLine($"subscription: {publishResponse.SubscriptionId}; clientHandle: {min.ClientHandle}; value: {min.Value}");
        }
    }

    publishRequest = new PublishRequest
    {
        SubscriptionAcknowledgements = new[] { new SubscriptionAcknowledgement { SequenceNumber = publishResponse.NotificationMessage.SequenceNumber, SubscriptionId = publishResponse.SubscriptionId } }
    };
}

Console.ReadKey(true);
```

BlackyPanther, 2017-04-19:
Hey Andrew,    
thank you for the details of the OPC-UA... it seems that I missed that while reading the specs.

Got another approach that now works for me using the client handle to map the value back to its node id.

closed 2017-04-19 by BlackyPanther