convertersystems/opc-ua-client #33: UaClient installation in .NET portable projects

abrasat, 2017-04-25:
I am getting the following error when trying to install the "Workstation.UaClient" NuGet package into a .NET portable project. What profile should be chosen for the .NET portable project to get it installed ?

...
Could not install package 'Workstation.UaClient 1.5.12'. You are trying to install this package into a project that targets '.NETPortable,Version=v4.5,Profile=Profile111', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.

awcullen, 2017-04-25:
I have no plans to support the portable library profiles.

awcullen, 2017-04-26:
That is a nice trick regarding directly referencing the .dll.  
Closing this issue for now.

closed 2017-04-26 by awcullen