convertersystems/opc-ua-client #34: issue in latest  release Nugets Workstation.UaClient - 1.5.12

tejasklodha, 2017-04-29:
[image]

I made Xamarin shared Mobile application in Android, iOS, UWP.
Application working fine with Nuget **Workstation.UaClient - 1.5.7**

When I Update Nuget to  **Workstation.UaClient - 1.5.12**  and rebuild same project then getting below errors in UWP application of xamarin 

As per attached screen shoot 

[image]

1. Error	CS0246	The type or namespace name 'LoggerFactory' could not be found 
2.  Error	Metadata	Error while reading ......#path#.......

Kindly help to resolve issue in stable release **Workstation.UaClient - 1.5.12** 

**Notes :**  In  **Workstation.UaClient - 1.5.12**  application working fine in Xamarin.Droid and Xamarin.iOS but issue is only in UWP application 

Thanks,

awcullen, 2017-04-29:
The new version changed a package dependency from 'Microsoft.Extensions.Logging' to Microsoft.Extensions.Logging.Abstractions'.  The reason being that I was only using the interfaces. 

In your Droid, iOS, and UWP projects you should add a package reference to 'Microsoft.Extensions.Logging.Debug' if you wish to create a LoggerFactory.

Sorry for the confusion.

tejasklodha, 2017-05-01:
Dear Sir,

Problem resolved when I install **Microsoft.Extensions.Logging** 

Thanks,

closed 2017-05-01 by tejasklodha