convertersystems/opc-ua-client #35: SubscriptionBase (opc-ua-client 2.0.0)

abrasat, 2017-05-19:
Usually I use a base view model class for mvvm applications (which implements IPropertyChanged, handles commands,...). The problem occurs when having to derive the application view model also from the new SubscriptionBase class, in view to get the opc-ua subscription working, as it is not possible to derive from two different classes. Is there any other way to subscribe to opc-ua items over the MonitoredItem attributes in this case ? Would be possible to add back the actual approach with the UaTcpSessionClient class over the "Subscribe()" method ?

awcullen, 2017-05-19:
Could your base view model class derive from SubscriptionBase?
Would you be able to show us your code?

abrasat, 2017-05-19:
This would be indeed possible when using a self-written base view model. But if using a third-party mvvm framework (lately I used for instance the Codon mvvm framework) it is not possible to modify it. I think that it is a restriction for the application to be forced to derive from SubscriptionBase.

awcullen, 2017-05-21:
There are a few ways to handle this. The easiest is to make a Subscription property of your ViewModel. Then update your bindings to bind to 'Subscription.property'.

For example:
```
    public class MainViewModel : Codon.UIModel.ViewModelBase
    {
        public MainSubscription Subscription { get; } = new MainSubscription();
    }

    [Subscription(endpointUrl: "opc.tcp://localhost:26543", publishingInterval: 500, keepAliveCount: 20)]
    public class MainSubscription : SubscriptionBase
    {
        ...
    }

    // In MainView.xaml
   <Slider Value="{Binding Subscription.Robot1Axis1}" />
```

closed 2017-06-05 by awcullen