convertersystems/opc-ua-client #36: Custom Structured Data Types

tomivirkkula, 2017-05-22:
What is the correct way to add custom data types to my project? For example, this STRUCT in my ST code:
```
TYPE MyStruct :
STRUCT
  a : INT;
  b : BOOL;
  c : STRING;
END_STRUCT
END_TYPE
```

My application sees the struct as a byte array and I managed to parse it in my MonitoredItem code, but is there a simpler and more correct way to do it?

awcullen, 2017-05-22:
I don't know of a way to add custom types to a commercial OPC Server, like Simatic.Net, Kepware, etc.

It is possible if you have written your own OPC Server, using an SDK like OPC Foundation's, UnifiedAutomation, etc. But I have not tried it myself.

On the client side then you would make a custom type that implements IEncodable, and then register it with the static method UaTcpSecureChannel.RegisterEncodable().
```
	[BinaryEncodingId("nsu=http://tomivirkkula.com/CustomTypes;s=MyStruct_BinaryEncoding")]
	[DataTypeId("nsu=http://tomivirkkula.com/CustomTypes;s=MyStruct_DataType")]
	public class MyStruct : IEncodable {
		public Int16 a { get; set; }
		public Boolean b { get; set; }
		public String c { get; set; }
        public virtual void Encode(IEncoder encoder) {
            encoder.PushNamespace("http://tomivirkkula.com/CustomTypes");
	    encoder.WriteInt16("a", a);
	    encoder.WriteBoolean("b", b);
	    encoder.WriteString("c", c);
            encoder.PopNamespace();
        }
        public virtual void Decode(IDecoder decoder) {
            decoder.PushNamespace("http://tomivirkkula.com/CustomTypes");
	    a= decoder.ReadInt16("a");
	    b= decoder.ReadBoolean("b");
	    c= decoder.ReadString("c");
            decoder.PopNamespace();
        }
	}
```

tomivirkkula, 2017-05-23:
Thanks for the quick reply. Now I got reading from the PLC working. But when I try to set the value in my MainViewModel I get a ArgumentOutOfRangeException "Type is unsupported". It seems the problem is that Variant doesn't know about MyStruct, since it looks for it in elemTypeMap, which is hard coded.

I think Variant should be seeing MyStruct as an ExtensionObject. Am I missing a cast  or something? Any help is greatly appreciated.

awcullen, 2017-05-23:
To write the MyStruct value you can wrap it in an ExtensionObject. You first have to find the NodeId that the server uses to send the variable.
```
new DataValue(new ExtensionObject(bytes, myStructTypeNodeId))
```

tomivirkkula, 2017-05-24:
I got writing working but I had to change my MonitoredItem to a DataValue instead of a MyStruct. Thank you once again!

awcullen, 2017-06-05:
You makes a good point. If you want to read or write the DataValue, you can change the signature of the property to:
```
        [MonitoredItem(nodeId: "ns=2;s=Robot1_Axis1")]
        public DataValue Robot1Axis1
        {
            get { return this.robot1Axis1; }
            set { this.SetProperty(ref this.robot1Axis1, value); }
        }

        private DataValue robot1Axis1;
```

closed 2017-06-05 by awcullen