convertersystems/opc-ua-client #37: Fetch historical data of OPC UA from PLC?

tejasklodha, 2017-05-31:
Dear Sir,

We want to access historical data of OPC UA from My PLC

**Practical scanario:**

One wpf application show live data from PLC throgh OPC UA.
somehow network connection break between PLC and my wpf application Now how can we face this issue?

can I fetch all historical data from PLC in view model?

I refer issue  https://github.com/convertersystems/opc-ua-client/issues/31

I try to run code as per attached file then unable to find historical data as below output.

[image]

[Program.zip](https://github.com/convertersystems/opc-ua-client/files/1040537/Program.zip)

Thanks

awcullen, 2017-05-31:
Storage of historical data and events is not a feature of most OPC UA servers today. Hopefully that might change in future. However, today there is a product from ProSysOpc that would meet your needs:
https://www.prosysopc.com/products/opc-ua-historian/

tejasklodha, 2017-05-31:
Dear Sir,

But I want to use in my code by using opc-ua-client.
Can it not possible from above code?

This is one of the most requirement in my current live project.

Thanks,

awcullen, 2017-05-31:
Your code is good. I would check if (StatusCode.IsGood(result.StatusCode) to avoid the null exception. 

The main problem is that the sample RobotServer does not store/support historical data or historical events.  Most OPC UA servers do not, at this time. An exception is the OPC-UA-Historian from ProSysOpc, which essentially adds centralized storage of historical data and events from your existing UA servers.

An opc-ua-client would be configured to connect to the ProSysOpc historian when getting data for trending, reporting or alarm history.

tejasklodha, 2017-05-31:
Dear sir,

Thanks for your response...
I will check our controller that, is it support to store historical data or not...

We are using b&r controller....

Thanks

tejasklodha, 2017-06-06:
Right now unable to fetch historical data from my controller....

closed 2017-06-15 by tejasklodha