convertersystems/opc-ua-client #40: Support for DiscreteAlarmType and LimitAlarmType

abrasat, 2017-07-13:
I could not find DiscreteAlarmType (and its subtype OffNormalAlarmType and so on) and LimitAlarmType (with subtype ExclusiveLimitAlarmType and so on) in the opc-ua-client ServiceModel, are these not supported yet ?

awcullen, 2017-07-14:
I stopped at AlarmCondition. If you like this feature, let me know. We can add more derived classes.

abrasat, 2017-07-14:
Yes, please add it. We have a "custom" opc-ua server which uses these types.

awcullen, 2017-07-15:
As I look at the OffNormalAlarmType and ExclusiveLimitAlarmType, I see that they do not _add_ any useful fields (compared to the AlarmConditionType) that would be displayed on an hmi. So I ask what purpose is the additional types?

If you use AlarmCondition as the property type, you still receive every alarm and event, from the base SystemEvent to the most customized Condition. Using AlarmCondition as the property type only specifies that the program receives the fields of the AlarmCondition. These are:
EventId
EventType
SourceName
Time
Message
Severity
ConditionId
ConditionName
BranchId
Retain 
AckedState
ConfirmedState
ActiveState

In the following example, the Robot1EventStream receives every event and alarm of the Robot1, and stores them in a queue (of the latest 16).
```
[MonitoredItem(nodeId: "ns=2;s=Robot1", attributeId: AttributeIds.EventNotifier)]
private AlarmCondition Robot1EventStream 
{ 
     set { this.Robot1Events.Enqueue(value); }
}

public ObservableQueue<AlarmCondition> Robot1Events { get; } = new ObservableQueue<AlarmCondition>(capacity: 16, isFixedSize: true);
```

abrasat, 2017-07-15:
Does the opc-ua client not receive from the server the information which is the "normal" defined value when OffNormalAlarmType occurs, respectively which of the LowLimit, LowLowLimit, HighLimit or HighHighLimit violation lead to the generation of ExclusiveLimitAlarmType ?

awcullen, 2017-07-15:
I have been using UaExpert and the UaServerCpp to test the alarms and events. With UaExpert, I connect to the server. I add an "Event view" document, browse to the "Demo.004_Events.AlarmsNoNodes" node and drag and drop the node onto the Configuration section of "Event View".  Then, to test the alarms, I write values to the NoNodesOffNormalAlarmTrigger and the NoNodesExclusiveLevelAlarmTrigger.

[image]

UaExpert chooses the most common fields to request, but you can use the UI to select additional fields.  By default, UaExpert does not select the NormalState, nor the HighHighState, LowLowState, etc.  I believe UaExpert expects the server to provide the info in the message field.

I followed UaExpert's advice when designing the Alarm classes, and I included only the most common fields.  There are literally 100's of other fields that could have been included. How to choose?

Happily, when using UaClient, you can make a customized Alarm class that exactly fits your needs. 
(see next message).

awcullen, 2017-07-15:
First, define a class that extends AlarmCondition. Add the fields that you wish to receive.
Add a Deserialize function which starts with field 13. (Alarm Condition uses fields 0-12)
Add a GetSelectClauses function that returns a SimpleAttributeOperand for each new field.
I used "LimitState/CurrentState/Id" because it is the NodeId of the active limit. The nodeId will match one of:

ObjectIds.ExclusiveLimitStateMachineType_HighHigh;
ObjectIds.ExclusiveLimitStateMachineType_High;
ObjectIds.ExclusiveLimitStateMachineType_Low;
ObjectIds.ExclusiveLimitStateMachineType_LowLow;

You could use "LimitState/CurrentState" which returns a localized text, eg "HighHigh". But then it could be in different language.
```
 /// <summary>
 /// Represents my customized alarm class.
 /// </summary>
 public class MyCustomAlarm : AlarmCondition
 {
     public NodeId LimitState { get; set; }

     public NodeId NormalState { get; set; }

     public override void Deserialize(Variant[] fields)
     {
         base.Deserialize(fields);
         this.LimitState = fields[13].GetValueOrDefault<NodeId>();
         this.NormalState = fields[14].GetValueOrDefault<NodeId>();
     }

     public override IEnumerable<SimpleAttributeOperand> GetSelectClauses()
     {
         foreach (var clause in base.GetSelectClauses())
         {
             yield return clause;
         }

         yield return new SimpleAttributeOperand { TypeDefinitionId = NodeId.Parse(ObjectTypeIds.ExclusiveLimitAlarmType), BrowsePath = new[] { new QualifiedName("LimitState"), new QualifiedName("CurrentState"), new QualifiedName("Id") }, AttributeId = AttributeIds.Value };
         yield return new SimpleAttributeOperand { TypeDefinitionId = NodeId.Parse(ObjectTypeIds.OffNormalAlarmType), BrowsePath = new[] { new QualifiedName("NormalState") }, AttributeId = AttributeIds.Value };
     }
 }
```

Then you can use MyCustomAlarm class in your view models.
```

[MonitoredItem(nodeId: "ns=2;s=Robot1", attributeId: AttributeIds.EventNotifier)] 
private MyCustomAlarm Robot1EventStream  
{       
     set { this.Robot1Events.Enqueue(value); } 
}
--

```

abrasat, 2017-07-16:
Thank you, I will try to implement custom alarms with our the custom opc-ua server.
How can the check of the EventType be integrated with the code above, to know if the NormalState (for OffNormalAlarmType) or the LimitState (for the ExclusiveLimitAlarmType) is relevant ?
Would the not-relevant fields just be null ?

awcullen, 2017-07-16:
If a field is not relevant then the field value will be the default for the field type. In the case of NormalState, the value will be null.

I have been testing with  these tools:
[OPC UA C++ Demo Server ](https://www.unified-automation.com/downloads/opc-ua-servers.html)
[UaExpert](https://www.unified-automation.com/downloads/opc-ua-clients.html)

abrasat, 2017-07-18:
Can alarms also be acknowledged or confirmed with the UaServerCpp demo ?
What is the expected behaviour after the opc-ua client calls the acknowledge (or confirm) method: the result of the call tells the opc-ua client that the alarm was ackowledged ? Or does the opc-ua server send again the alarm condition with the AckedState set ?

abrasat, 2017-07-18:
After the type of AckedState, ConfirmedState was changed to bool (version 2.0.1), they are set to false, even if the opc-ua server does not send these fields.
In the previous opc-ua client version (2.0.0) these were null when not available.

awcullen, 2017-07-18:
Did you find out if the opc-ua server sends again the alarm condition with the AckedState set?
Does the change to bool cause problems with your application?
I could change the property type to bool? (nullable), which can be null when the server does not send the field.

abrasat, 2017-07-19:
As far as I can see with UaServerCpp+UaExpert, the opc-ua server sends for ExclusiveLevelAlarm indeed again the alarm condition with the AckedState=True after acknowledging the alarm (AckedState=false when the alarm occurs). On the other hand, the OffNormal alarm has AckedState set to true when the alarm occurs, probably the UaServerCpp signals that it does not need an Acknowledge for that. I will test later also with opc-ua client and the customized alarm.

But anyway, from the opc-ua client point of view I think that a nullable bool for AckedState, ConfirmedState would be a better option

How exactly can UaExpert be configured to show also the LimitState or the NormalState ?

.

awcullen, 2017-07-20:
UaExpert allows you to select the fields to receive in the top "Configuration" area (see above).
The additional fields appear in the "Details" panel.
It seems that LimitState is reported properly with the ExclusiveLevelAlarm type. But I did not get a NormalState from an OffNormalAlarm.

awcullen, 2017-07-21:
Nuget V2.0.2 has added helpful methods to SubscriptionBase:
```

public async Task<StatusCode> Acknowledge(AcknowledgeableCondition condition, LocalizedText comment = null);

public async Task<StatusCode> ConditionRefresh()

```
Also updated BaseEvent and subclasses, Added EventField attribute to properties to better control the generation of SelectClauses and deserialization. Please update your custom alarm classes.

abrasat, 2017-07-22:
Sounds great, I will test it on Monday.
Are the last changes committed here on github ?

awcullen, 2017-07-23:
Thanks, Forgot to Push.

abrasat, 2017-07-24:
Regarding the custom alarm classes, is it still necessary to override the "Deserialize()" and "GetSelectClauses()" methods, or only add event field attributes ?

awcullen, 2017-07-24:
Only add event field attributes.

closed 2017-08-23 by awcullen