convertersystems/opc-ua-client #43: Array change event not Working Properly

tejasklodha, 2017-08-03:
Dear Sir,

I created one OPC array which length is approx 400.
Now I Observe that when my one value of array is changed then my array still not updated.
Its happen sometimes only....
Am I missing something?
Below is my array defin in view model

` [MonitoredItem(nodeId: "ns=6;s=::AsGlobalPV:ActData_OPC")]
        public UInt32[] Global_PVActData_OPC
        {
            get { return this.global_PVActData_OPC; }

            set
            {
                this.SetProperty(ref global_PVActData_OPC, value);
             }
        }`

awcullen, 2017-08-03:
The problem is that the property doesn't know when the user changes an element of the array.
What I do is this:

```
// take copy of array
var a = sub.Global_PVActData_OPC.ToArray();
// change an element of copy
a[2] = a[2] + 1;
// set property with copy
sub.Global_PVActData_OPC = a;

```

Also, would you tell us if the new UaBrowser is working with the "GlobalPv" folder.
see your issue: https://github.com/convertersystems/opc-ua-tools/issues/4

tejasklodha, 2017-08-07:
Ok....

Problem resolved...

awcullen, 2017-08-07:
Good.

closed 2017-08-07 by awcullen