convertersystems/opc-ua-client #45: Opc ua client logging with Microsoft.Extensions.Logging.Abstractions

abrasat, 2017-08-30:
The dependency of the opc-ua-client on Microsoft.Extensions.Logging.Abstractions makes it difficult to use it in a Xamarin.Forms shared project for the iOS target, as the ILoggerFactory from Microsoft.Extensions.Logging.Abstractions seems to be missing from Xamarin.Ios.
Could you please provide a opc-ua-client sample for Android, iOS and UWP, with the possibility to define differrent loggers for the opc-ua-client for each platform ?
What Microsoft.Extensions.Logging providers are actually available for Android, iOS and UWP ?

awcullen, 2017-08-31:
I see that you asked a question on the Xamarin Forms forum. Hopefully there will be a response soon.
Sometimes I use the logging provider of [Serilog](https://github.com/serilog/serilog-extensions-logging).

abrasat, 2017-08-31:
The problem is that the Microsoft.Extensions.Logging.Abstractions dependency prevents the usage of opc-ua-client in a shared project in iOS projects. I will try to put the code using the opc-ua-client in a PCL project (as far that I saw the opc-ua-client nuget package does not install correctly in a PCL project). Are there any opc-ua-client samples available for PCL projects ?

awcullen, 2017-08-31:
I have never built a Xamarin iOS project, before.  I created a new Xamarin project to test. In the iOS project I switched the packages.config to a project.json,  
```
{
  "dependencies": {
    "Microsoft.Extensions.Logging": "1.1.2",
    "Microsoft.Extensions.Logging.Debug": "1.1.2",
    "NETStandard.Library": "1.6.1",
    "StyleCop.Analyzers": "1.0.2",
    "Workstation.UaClient": "2.0.2",
    "Xamarin.Forms": "2.3.4.247"
  },
  "frameworks": {
    "Xamarin.iOS,Version=v1.0": {}
  },
  "runtimes": {
    "win": {}
  }
}
```

The project built without errors, but I do not have a way to test it at the moment. (No Mac)

What setup do you use for compiling/testing a Xamarin.iOS project. A Mac? MacInCloud?

abrasat, 2017-08-31:
I had a Mac for tests.  Will have it again next week, will take then a look.

awcullen, 2017-09-21:
I added a MobileHmi.iOS project to the samples. Would you test it and let me know the result?

abrasat, 2017-09-28:
I could not compile the iOS project, the following error occurs : Failed to resolve "Microsoft.IO.RecyclableMemoryStreamManager" reference from "Microsoft.IO.RecyclableMemoryStream, Version=1.2.2.0,...

Not sure where that comes from.

For opc-ua client test purposes, I think it would be useful to add also a UWP version of Workstation.MobileHmi.

closed 2019-07-28 by awcullen