convertersystems/opc-ua-client #47: How to set a subscription interval

GengHe, 2017-09-12:
In the opc ua server,the tag of the change cycle is one second. In the program `var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);`  , I need to wait four seconds to read the data. My code is as follows:
`//Create a subscription
            var subscriptionRequest = new CreateSubscriptionRequest
            {
                RequestedPublishingInterval = 500,
                RequestedMaxKeepAliveCount = 10,
                RequestedLifetimeCount = 30,
                PublishingEnabled = true
            };
            var subscriptionResponse = await session.CreateSubscriptionAsync(subscriptionRequest).ConfigureAwait(false);
            var id = subscriptionResponse.SubscriptionId;
            //Add items to the subscription
            var itemsToCreate = new MonitoredItemCreateRequest[]
               {
                    new MonitoredItemCreateRequest
                    {
                        ItemToMonitor = new ReadValueId
                        {
                            NodeId = NodeId.Parse("ns=2;s=Channel1.Device1.Tag1"),
                            AttributeId = AttributeIds.Value
                        },
                        MonitoringMode = MonitoringMode.Reporting,
                        RequestedParameters = new MonitoringParameters
                        {
                            ClientHandle = 1201,
                            SamplingInterval = -1,
                            QueueSize = 0,
                            DiscardOldest = true
                        }
                    }
               };
            var itemsRequest = new CreateMonitoredItemsRequest
            {
                SubscriptionId = id,
                ItemsToCreate = itemsToCreate,
            };
            var itemsResponse = await session.CreateMonitoredItemsAsync(itemsRequest).ConfigureAwait(false);
            //Publish the subscription
            var publishRequest = new PublishRequest
            {
                SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0]
            };
            while (true)
            {
                try
                {
                    var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);
                    var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                    foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            if (this.OpcEvent != null)
                            {
                                OpcEvent(min.Value.ToString(), new Tag { Path = "123", TagName = "test" });
                            }

                            //Console.WriteLine($"subscription: {publishResponse.SubscriptionId}; clientHandle: {min.ClientHandle}; value: {min.Value}");
                        }
                    }

                    publishRequest = new PublishRequest
                    {
                        SubscriptionAcknowledgements = new[] { new SubscriptionAcknowledgement { SequenceNumber = publishResponse.NotificationMessage.SequenceNumber, SubscriptionId = publishResponse.SubscriptionId } }
                    };
                }
                catch
                {

                    
                }
               
            }`
 I would like to know where to set the subscription interval.

awcullen, 2017-09-12:
Hello GengHe,
You have a choice. You can change the item's SamplingInterval to 4000. This tells the server to sample the dataValue every 4 seconds. Easy!

Or you can change the subscription's RequestedPublishingInterval  to 4000. This tells the server to store the datachanges in a queue and return them all in the next scheduled PublishResponse (which is returned no faster than the 4000 ms). In your code above, the item's QueueSize is zero, so in fact the stored datachanges are discarded, so you will receive the only latest value every 4 seconds.

But most importantly, if you have installed a version of the Workstation.UaClient > 2.0 then you will need to change some of your code. In the new version, the UaTcpSessionChannel publishes the subscription automatically.  You just need to call Subscribe with a method that handles the PublishResponse.
 
At the point in your code where you start a loop to publish the subscription, replace your code with 

```
// Publish the subscription
var token = session.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
{
    // loop thru all the data change notifications
    var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
    foreach (var dcn in dcns)
    {
        foreach (var min in dcn.MonitoredItems)
        {
            if (this.OpcEvent != null)
            {
                OpcEvent(min.Value.ToString(), new Tag { Path = "123", TagName = "test" });
            }
            //Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
        }
    }
});

```

GengHe, 2017-09-13:
Thank you very much for your help, my problem has been solved.

closed 2017-09-13 by GengHe