convertersystems/opc-ua-client #48: Missing events in client opc ua client

basprins, 2017-09-14:
Hi Andrew,

In our software (client) we miss events from the B&R PLC (server). The PLC task is running in a 2ms cycle time. The subscription is running with the following construction:

            var subscriptionRequest = new CreateSubscriptionRequest
            {
                //* Below comments are copied from https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/ *//
                //* A different library, but still a very good explanation of what these properties mean *//

                /**
                 * PublishingInterval defines how often the server checks if there are notification
                 * packages (publish responses) for a Subscription to be sent back to the client.
                 * A typical value is 1000 (ms). It should not be -1 nor 0.
                 * Instead of using very small PublishingInterval, consider using a smaller SamplingInterval.
                 * The server may also increase the PublishingInterval to a default minimum.
                 *
                 * NOTE: we do not follow above guideline since intervals of 1 second on changes will impact our performance dramatically.
                 */
                RequestedPublishingInterval = 1,

                /**
                 * If there is no data to send after the next PublishingInterval, the server will skip it.
                 * But KeepAlive defines how many intervals may be skipped, before an empty notification is
                 * sent anyway: to give the client a hint that the subscription is still alive in the server
                 * and that there just has not been any data arriving to the client.
                 */
                RequestedMaxKeepAliveCount = 100,

                /**
                 * The client’s responsibility is to send PublishRequests to the server, in order to enable the
                 * server to send PublishResponses back. The PublishResponses are used to deliver the
                 * notifications: but if there are no PublishRequests, the server cannot send a notification
                 * to the client.
                 *
                 * The server will also verify that the client is alive by checking that new
                 * PublishRequests are received – LifeTimeCount defines the number of PublishingIntervals to wait
                 * for a new PublishRequest, before realizing that the client is no longer active. The Subscription is then removed from the server.
                 */
                RequestedLifetimeCount = 10*60*60, // RequestedPublishingInterval * 10 * 60 * 60 = 1 hour
                PublishingEnabled = true,
            };

            var subscriptionResponse = session.CreateSubscriptionAsync(subscriptionRequest).Result;
            var sid = subscriptionResponse.SubscriptionId;

            value = ReadValue();
            var itemsToCreate = new[]
            {
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor = new ReadValueId
                    {
                        NodeId = nodeId,
                        AttributeId = AttributeIds.Value
                    },
                    MonitoringMode = MonitoringMode.Reporting,
                    RequestedParameters =
                        new MonitoringParameters
                        {
                            ClientHandle = clientHandle,
                            SamplingInterval = 1,
                            QueueSize = 5,
                            DiscardOldest = true
                        }
                }
            };

            var itemsRequest = new CreateMonitoredItemsRequest
            {
                SubscriptionId = sid,
                ItemsToCreate = itemsToCreate,
            };
            session.CreateMonitoredItemsAsync(itemsRequest).Wait();

In the scenario that fails the property we are monitoring changes from one value to another and back to the first value in three consecutive cycles (cycle N: 13, cycle N+1: 2, cycle N+2: 13). In the client application I log every change in monitored items, but the change from 13 to 2 and back to 13 go unnoticed. 

Can I improve this by using a bigger QueueSize? I tried to use QueueSize=5 but that did not change anything. Or should I send publish requests / monitor items in a different way as posted below?

        public async Task StartMonitoring()
        {
            var publishRequest = new PublishRequest
            {
                SubscriptionAcknowledgements = new SubscriptionAcknowledgement[0],
                RequestHeader = new RequestHeader
                {
                    TimeoutHint = 1000*60*60
                }
            };

            while (true)
            {
                try
                {
                    var publishResponse = await session.PublishAsync(publishRequest).ConfigureAwait(false);

                    // loop thru all the data change notifications
                    var dcns = publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                    foreach (var dcn in dcns)
                    {
                        foreach (var min in dcn.MonitoredItems)
                        {
                            var prop = monitoredOpcUaProperties.Single(p => p.ClientHandle == min.ClientHandle);
                            prop.UpdateValue(min.Value.Value);
                        }
                    }

                    publishRequest = new PublishRequest
                    {
                        SubscriptionAcknowledgements = new[]
                        {
                            new SubscriptionAcknowledgement
                            {

                                SequenceNumber = publishResponse.NotificationMessage.SequenceNumber,
                                SubscriptionId = publishResponse.SubscriptionId
                            }
                        }
                    };
                }
                catch (ServiceResultException e)
                {
                    if (!e.Message.Equals("Timeout occurred while processing the request."))
                    {
                        throw;
                    }

                    // If publish request/response was lost, try to republish
                    for (var i = 1; i < RetryCounter; i++)
                    {
                        try
                        {
                            var republishRequest = new RepublishRequest();
                            var publishResponse = await session.RepublishAsync(republishRequest).ConfigureAwait(false);

                            // loop thru all the data change notifications
                            var dcns =
                                publishResponse.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                            foreach (var dcn in dcns)
                            {
                                foreach (var min in dcn.MonitoredItems)
                                {
                                    var prop = monitoredOpcUaProperties.Single(p => p.ClientHandle == min.ClientHandle);
                                    prop.UpdateValue(min.Value.Value);
                                }
                            }
                        }
                        catch (ServiceResultException ex)
                        {
                            if (i >= RetryCounter)
                            {
                                throw new OpcUaCommunicationException($"Unable to repulish, retried '{i}' times", ex);
                            }
                        }
                    }
                }
            }
            // ReSharper disable once FunctionNeverReturns
        }

Really hope you can help me out once again.

Many thanks in advance.

awcullen, 2017-09-14:
Hi,

In Workstation.UaClient version 2.0.0 and higher, the UaTcpSessionChannel starts publishing automatically. Now you call Subcribe() with a method that handles the PublishResponse. Below I modified some of your code to show what I mean.

```
var subscriptionRequest = new CreateSubscriptionRequest
{
    /**
     * PublishingInterval defines how often the server checks if there are notification
     * packages (publish responses) for a Subscription to be sent back to the client.
     * A typical value is 1000 (ms). It should not be -1 nor 0.
     * Instead of using very small PublishingInterval, consider using a smaller SamplingInterval.
     * The server may also increase the PublishingInterval to a default minimum.
     *
     * NOTE: we do not follow above guideline since intervals of 1 second on changes will impact our performance dramatically.
     */
    RequestedPublishingInterval = 1,

    /**
     * If there is no data to send after the next PublishingInterval, the server will skip it.
     * But KeepAlive defines how many intervals may be skipped, before an empty notification is
     * sent anyway: to give the client a hint that the subscription is still alive in the server
     * and that there just has not been any data arriving to the client.
     */
    RequestedMaxKeepAliveCount = 100,

    /**
     * The client’s responsibility is to send PublishRequests to the server, in order to enable the
     * server to send PublishResponses back. The PublishResponses are used to deliver the
     * notifications: but if there are no PublishRequests, the server cannot send a notification
     * to the client.
     *
     * The server will also verify that the client is alive by checking that new
     * PublishRequests are received – LifeTimeCount defines the number of PublishingIntervals to wait
     * for a new PublishRequest, before realizing that the client is no longer active. The Subscription is then removed from the server.
     */
    RequestedLifetimeCount = 1 * 1000 * 60 * 60, // 1 * 1000 * 60 * 60 = 1 hour // check your math!
    PublishingEnabled = true,
};

var subscriptionResponse = session.CreateSubscriptionAsync(subscriptionRequest).Result;
var sid = subscriptionResponse.SubscriptionId;
var revisedPublishingInterval = subscriptionResponse.RevisedPublishingInterval; // fyi - the server is allowed to revise the interval

value = ReadValue();
var itemsToCreate = new[]
{
    new MonitoredItemCreateRequest
    {
        ItemToMonitor = new ReadValueId
        {
            NodeId = nodeId,
            AttributeId = AttributeIds.Value
        },
        MonitoringMode = MonitoringMode.Reporting,
        RequestedParameters =
            new MonitoringParameters
            {
                ClientHandle = clientHandle,
                SamplingInterval = -1, // -1 means as fast as possible
                QueueSize = 5,
                DiscardOldest = true
            }
    }
};

var itemsRequest = new CreateMonitoredItemsRequest
{
    SubscriptionId = sid,
    ItemsToCreate = itemsToCreate,
};
var itemsResult = session.CreateMonitoredItemsAsync(itemsRequest).Result;
var revisedSamplingInterval = itemsResult.Results[0].RevisedSamplingInterval; // fyi - the server is allowed to revise the interval

var token = session.Where(pr => pr.SubscriptionId == sid).Subscribe(pr =>
{
    // loop thru all the data change notifications
    var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
    foreach (var dcn in dcns)
    {
        foreach (var min in dcn.MonitoredItems)
        {
            var prop = monitoredOpcUaProperties.Single(p => p.ClientHandle == min.ClientHandle);
            prop.UpdateValue(min.Value.Value);
        }
    }
});

```

Also note how the server responses return revised values for PublishingInterval and SamplingInterval. The B&R server may adjust your requested intervals to something like 50 ms. 

Which brings me to my last point.  I don't see how any OpcServer will be able to scan the PLC memory every 1 ms.  I think you will have to change the PLC program. There are lots of options to signal an event to a HMI: 
 An Off- Delay timer (TOF) with a delay that is 2x the sampling interval.

 A counter that increments with each event, rolling over to zero.
i  := i +1 MOD 100;   

Andrew

basprins, 2017-09-17:
Hi Andrew,

Thanks for all the additional info! Will upgrade to latest version (I already tried before posting, but ran into breaking changes and didn't have the time to adapt). Also good to learn that we are pushing the limits of OpcUa with our 1ms approach. We were always wondering to what extend that would really hold, I guess up until this point then :).

We'll have to revise some of the choices we made. Maybe we will try to lean on you more if we can't find answers elsewhere. Our problem is that we are no OpcUa experts, and have to learn by reading documentation which isn't always easy to find on specific topics...

Anyway, thanks a bunch once again. 

PS: What is the problem with the math? 
RequestedLifetimeCount = 1 * 1000 * 60 * 60, // 1 * 1000 * 60 * 60 = 1 hour // check your math!
1 hour = 3600000 ms
Or am I missing the obvious here?

awcullen, 2017-09-17:
I am curious what your server returns as RevisedPublishingInterval and RevisedSamplingInterval.

I saw that the original code at the top of the post was: 
RequestedLifetimeCount = 10*60*60

basprins, 2017-09-19:
OK, here goes... 

I have two branches of *our* software:
- one version with ua-client 2.0.2
- one version with ua-client 1.5.12

For the rest I changed nothing in our legacy application, so that the necessary changes to cope with the breaking interface changes are done, and I added some loggings to compare the two versions.

Thus I also use the same subscription defintions:
- RequestedPublishingInterval (in the CreateSubscriptionRequest) = 50 
- SamplingInterval (in the MonitoringParameters)= -1 as suggested by you earlier 
- QueueSize (in the MonitoringParameters) = 100, an arbitrary size, more than 1 what I had earlier

In the 1.5.12 version I can repeatedly start the software, and start production => the amount of missing state changes is exceptional (but that's not good enough, I need to be able to trust that OpcUa will communicate ALL changes of certain parameters, and the reason I started this thread on your forum).

In the 2.0.2 version I miss state changes, much more frequently (!). So the latest version seems to react differently when it comes to notifying changes over OpcUa.  

A potential Interesting note (and the answer to something you asked me to check):

Revised intervals for 1.5.12 client:
(from our loggings)
2017-09-19 13:08:31,202 [DEBUG][thread: UI][OpcuaClient] revisedPublishingInterval=50
2017-09-19 13:08:31,212 [DEBUG][thread: UI][OpcuaClient] revisedSamplingInterval=50

Revised intervals for 2.0.2 client:
(from our loggings)
	2017-09-19 10:42:10,274 [DEBUG][thread: UI][OpcuaClient] revisedSamplingInterval=10
	2017-09-19 10:42:10,277 [DEBUG][thread: UI][OpcuaClient] revisedPublishingInterval=50

Isn't it weird that the usage of the client results in a different revised sampling interval? I know it would be handy to isolate the client, so that we can cancel out any behaviour of our software. That would be a better test, I agree, but for me not easy to setup.

Regardless of all the above, my biggest concern remains that I miss still miss state changes, no matter which version I use. I am well aware of the possibility that this has nothing to do with your software (ua-client), and the OPC server in the PLC _could_ be solely responsible for this. 

I just hope that you can give me pointers to move forward with this. The OpcUa interface has a crucial place in our architecture. The news that I am not able to "trust" that an OPC server captures all changes for certain parameters would be rather devastating for our architecture. Or maybe in other words: the news that there is no way I can influence either the PLC or the OPC server, so that all relevant changes are captured and notified over OpcUa, would be rather devastating for our architecture. 

I don't mind  to try another OpcUa client (although I doubt that will give me a lot of new insights, let alone a working software stack).

Just to clarify what we are trying to achieve:
- The PLC runs a bunch of tasks with some clock cycle (2ms, some in 4ms, some in 6ms, not really of interest, they all have certain parameters which must be communicated to our C# application).
- The C# application "control" the PLC on high level tasks. It also is responsible for managing the entire administration of jobs and products. It can only do that job properly, if it can orchestrate the PLC on high level commands, like pick a product from X, and put it in track on position Y. The interface between C# and PLC is defined by the PLC state machine, so the C# application will wait for certain states. If in some scenarios these state changes go really fast, they tend to go unnoticed. Important note, we NEVER can change a state twice within one PLC cycle. We thought that this "design rule" would be sufficient for the OPC server to at least notice the change, and report it (in the worst case, delayed, but never miss the event). This last part seems to fail us know.

I hope:
- that you can make sense of all I am saying
- that you can find the time to share some tips/experience/guidelines

I hope that you can make sense of all I am saying and that you can help me further.

Many thanks in advance

Best regards, Bas

awcullen, 2017-09-19:
I searched at https://www.br-automation.com for any documentation of Opc UA server configuration, but did not find anything.

You have reported the revised sampling rate of 10 or 50 ms, depending on client. In either case that is too slow to reliably detect a data change that is reset after 2 to 6 ms.

I believe you will have to alter your PLC program to accomplish what you want.

You describe the value changing from 13 -> 2 -> 13.  What does the value represent? Is it important to capture all the values (dataChange), or just the fact that it changed (event)?

Andrew

P.S. The big change between version 1.5 and 2.0 is that the new version calls PublishAsync internally and reports the PublishResponses to delegates that are passed to Subscribe().

So, in version 2.0, you should not be calling PublishAsync yourself. No while(true) loop!

basprins, 2017-09-20:
Hi Andrew,

The 10ms and 50ms revised intervals were reported by the "B&R simulation PLC". I will also test on the actual machine what the real PLC reports back. Not that this really solves my problem, because it has to work in simulation as well of course. In any event, this is helpful information, I will report that back to B&R as well so that they can advise from their side as well.

The 13-> 2 -> 13 changes are enum values of different states. So in the case I miss a state change. So yes, it's mandatory in our case to capture all changed values.

By the way, these revised timings (10ms and 50ms), are they worst case numbers? Could it still be the case that the PLC has a faster sampling interval and/or faster publish interval? Because if it would really be sampling of 10ms then I would expect that the performance would be even worse.

PS: your subscribe method was clear, and also the huge benefit of getting rid of the while(true) loop! :). Great job on that part.

awcullen, 2017-09-20:
I am curious what B&R will say.
I would try slowing the plc program cycle to allow opc communications have greater time slice.

Regarding  capturing (13-> 2 -> 13):
Consider adding a timer that signals State2 to opc client.
```
"State2_DB".TOF(IN := "State" == 2, PT := "State_PresetTime");
```            
Then monitor item "State2_DB".Q

basprins, 2017-09-21:
I also tested on the real PLC, the same timings are reported back there. Both revised intervals are 50ms (!!). I will ask what this means in their OPC server.  B&R is coming over next week so I will let you know next week what came out.

basprins, 2017-09-25:
Hi Andrew, I will close this issue. B&R is going to dive deeper. They also mentioned an alternative way of communicating, starting to use OpcUa Methods. Something we already knew about but never really investigated. Not sure if it's even possible using your client. Anyway, I asked if it would be possible to first "be sure" that this path is a dead end before we "overhaul" our software. So it's on hold waiting for more info from B&R. 

Once again thanks for your time and help! B&R will also start using your client in order to reproduce the issues. Maybe that will give your client a bit more exposure 👍

closed 2017-09-25 by basprins

basprins, 2017-10-18:
Hi Andrew,

A final update on this issue.

First of all, the B&R PLC had some configuration after all where we could influence timers (sampling/publishing). We run on a publish interval of 50ms now, and sample interval of 10ms. A higher frequency of sampling was possible to define (e.g. 2ms) but when we go higher than 10ms the PLC crashes. They are looking into this at their development department atm.

In order to workaround this limitation, we changed the code in the PLC and implemented a minimum state transition duration. If a state change would occur faster then 10ms, we delay it so that the OpcUa server will at least capture the state change. A simple TON timer takes care of that. This works fine for now.

Thanks again for the help and support from your side!

awcullen, 2017-10-18:
Great news.  Ability to sample data at 10ms is a truly useful achievement. Congrats to B&R engineering.