convertersystems/opc-ua-client #50: EndPoint attributes for SubscriptionBase

udoliess, 2017-10-17:
I was looking for the best client design pattern to help me with automatic reconnects and subscriptions. So I tried to adapt your StatusHmi sample but it does not connect to OpenOpcUa.org demo server. As far as I understand, the client uses wrong endpoint instance. Here come my comments/questions:
- For class SubscriptionBase the endpoint can only be specified as string parameter of SubscriptionAttribute.
- In method UaApplication.CreateChannelAsync an instance of EndpointDescription is created by `endpoint = new EndpointDescription { EndpointUrl = endpointUrl };`.
- This EndpointDescription is used later in method UaTcpSessionChannel.UaTcpSessionChannel. The comments for this method say: "The EndpointDescription of the remote application. Obtained from a prior call to UaTcpDiscoveryClient.GetEndpoints.". But used endpoint was not obtained from UaTcpDiscoveryClient.GetEndpoints.
- In my case it looks like SecurityLevel does not fit and therefore no connection can be established.
- How can I specify a complete EndpointDescription with SecurityPolicyUri and SecurityLevel as attributes of SubscriptionBase?
- Or what else should I do to connect SubscriptionBase to OpenOpcUa.org demo server?

Thanks,
Udo

awcullen, 2017-10-17:
Hello Udo,

By default, the UaApplication will select the Endpoint with the highest SecurityLevel. Maybe you want the endpoint with no security?

The UaApplicationBuilder has a method to add a MappedEndpoint, where you can replace one Endpoint with another. In the following example, when a Subscription asks for endpoint "opc.tcp://localhost:26543", the UaApplication will substitute "opc.tcp://localhost:26543", SecurityPolicyUris.None

```
            this.application = new UaApplicationBuilder()
                .SetApplicationUri($"urn:{Dns.GetHostName()}:Workstation.StatusHmi")
                .SetDirectoryStore(Path.Combine(
                    Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData),
                    "Workstation.StatusHmi",
                    "pki"))
                .SetIdentity(this.ShowSignInDialog)
                .ConfigureLoggerFactory(o => o.AddDebug(LogLevel.Trace))
                .AddMappedEndpoint("opc.tcp://localhost:26543", "opc.tcp://localhost:26543", SecurityPolicyUris.None)
                .Build();
```

udoliess, 2017-10-19:
Hello Andrew,
Thank you very much for your answer. Your solution of adding an endpoint map helped in my case.

With this solution I tried to use UaApplication and SubscriptionBase in a minimal console program and found a trick to start the background activities: I had to install an even handler for PropertyChanged. Is this behavior intended?
```c#
using System;
using System.ComponentModel;
using System.Net;
using System.Threading;
using Workstation.ServiceModel.Ua;
[Subscription(endpointUrl: "opc.tcp://localhost:23562", publishingInterval: 50)]
class Program : SubscriptionBase
{
	static void Main(string[] args)
	{
		using (var app = new UaApplicationBuilder()
			.SetApplicationUri($"urn:{Dns.GetHostName()}:Test")
			.AddMappedEndpoint("opc.tcp://localhost:23562", "opc.tcp://localhost:23562", SecurityPolicyUris.None)
			.Build())
		{
			app.Run();
			var prog = new Program();
			PropertyChangedEventHandler dummy = delegate { };
			try
			{
				prog.PropertyChanged += dummy; // needed to start background activities
				while (!Console.KeyAvailable)
					Thread.Sleep(1);
			}
			finally { prog.PropertyChanged -= dummy; }
		}
	}
	[MonitoredItem(nodeId: "ns=2;i=203")]
	public float Val
	{
		get { return this.val; }
		private set { SetProperty<float>(ref val, value); Console.WriteLine(Val); }
	}
	private float val;
}
```
Thanks,
Udo

awcullen, 2017-10-19:
That's true that a Subscription needs to be observed to start connecting.

closed 2019-07-28 by awcullen