convertersystems/opc-ua-client #52: opc-ua-client in netstandard class library

abrasat, 2017-10-24:
Is it possible to use the opc-ua-client in a netstandard class library ? As I wrote in the xamarin forum, I have problems starting an Android application in the google android emulator, when this application references a netstandard dll containing the opc-ua-client.

abrasat, 2017-10-26:
I made some tests with the opc-ua client in a netstandard 2.0 class library.

When compiling the iOS application from a Xamarin.Forms project, which references the netstandard 2.0 library where the opc-ua client nuget package is installed,  the following error is generated:
```
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\iOS\Xamarin.iOS.Common.targets(777,3): error : Failed to resolve "Workstation.ServiceModel.Ua.NodeId" reference from "Workstation.UaClient, Version=2.1.1.0, Culture=neutral, PublicKeyToken=a5b9a5aa6d3e280f"
```

When compiling the android application from a Xamarin.Forms project the following error is generated:
```
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1445,2): error : Exception while loading assemblies: System.IO.FileNotFoundException: Could not load assembly 'Workstation.UaClient, Version=2.1.1.0, Culture=neutral, PublicKeyToken=a5b9a5aa6d3e280f'. Perhaps it doesn't exist in the Mono for Android profile?
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1445,2): error : File name: 'Workstation.UaClient.dll'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1445,2): error :    at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve(AssemblyNameReference reference, ReaderParameters parameters)
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1445,2): error :    at Xamarin.Android.Tasks.ResolveAssemblies.AddAssemblyReferences(DirectoryAssemblyResolver resolver, ICollection`1 assemblies, AssemblyDefinition assembly, Boolean topLevel)
C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets(1445,2): error :    at Xamarin.Android.Tasks.ResolveAssemblies.Execute(DirectoryAssemblyResolver resolver)
```

I cannot test the UWP application (yet), as the Windows 10 Fall Creators Update is not yet installed on my notebook.

abrasat, 2017-11-02:
After installation of  Windows 10 Fall Creators Update, update of Visual Studio 2017 to latest release and creation of test project from scratch, all three platforms can be compiled without errors.

closed 2017-11-02 by abrasat

awcullen, 2017-11-02:
I need to do the same.  
Did you find the UaClient for NetStandard 2.0 works in UWP with Fall Creators Update?
Should I drop support for NetStandard 1.4?

abrasat, 2017-11-13:
Yes, UaClient for NetStandard 2.0 works in UWP with Fall Creators Update.
Not sure how many users migrated to NetStandard 2.0 from NetStandard 1.4, maybe you should support both versions for a while.
But definitely NetStandard 2.0 (and higher) is the way to go.