convertersystems/opc-ua-client #55: opcua session was closed by client

balamcsd, 2017-11-24:
I have written the attached OpcUaConnector class for opc ua connection related activities.

But it is not handling session. For example :

1) in opc ua configuration disabled the endpoint 
2) in kepserver configuration did runtime > reinitializing

The windows service is throwing : 
Source : system.Reactive.Core 
InnerException : The session was closed by client

and stopping the windows service, as this error goes unhandled.

Can some one suggest how to handle session in opc-ua?
[OpcUaConnector.txt](https://github.com/convertersystems/opc-ua-client/files/1500882/OpcUaConnector.txt)

awcullen, 2017-11-26:
When a server closes the session and socket (as happens when you reinitialize Kepware) the client receives immediate notification that causes the client channel to fault. A faulted channel cannot be reopened, it should be aborted and a new channel should be created.

I made this standalone test, to show that you may have to catch an exception and recreate the channel and subscription. The point of this test is to subscribe to the CurrentTime node and collect 60 datachanges. The test lasts a minute.  If you Reinit the Kepware server, the test catches the exception and recreates the channel and subscription. 
```
        [TestMethod]
        public async Task OpcConnectorTest()
        {
            var count = 0;
            UaTcpSessionChannel channel = null;

            while (count < 60)
            {
                try
                {
                    channel = new UaTcpSessionChannel(
                            this.localDescription,
                            this.certificateStore,
                            new AnonymousIdentity(),
                            EndpointUrl,
                            SecurityPolicyUris.None,
                            loggerFactory: this.loggerFactory);

                    await channel.OpenAsync();

                    // create the keep alive subscription.
                    var subscriptionRequest = new CreateSubscriptionRequest
                    {
                        RequestedPublishingInterval = 1000f,
                        RequestedMaxKeepAliveCount = 30,
                        RequestedLifetimeCount = 30 * 3,
                        PublishingEnabled = true,
                    };
                    var subscriptionResponse = await channel.CreateSubscriptionAsync(subscriptionRequest).ConfigureAwait(false);
                    var id = subscriptionResponse.SubscriptionId;

                    var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
                    {
                        // loop thru all the data change notifications
                        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                        foreach (var dcn in dcns)
                        {
                            foreach (var min in dcn.MonitoredItems)
                            {
                                Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
                                count++;
                            }
                        }

                    });

                    var itemsRequest = new CreateMonitoredItemsRequest
                    {
                        SubscriptionId = id,
                        ItemsToCreate = new MonitoredItemCreateRequest[]
                        {
                            new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2258"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }
                        },
                    };
                    var itemsResponse = await channel.CreateMonitoredItemsAsync(itemsRequest);

                    while (channel.State == CommunicationState.Opened && count < 60)
                    {
                        await Task.Delay(1000);
                    }
                }
                catch (Exception ex)
                {
                    Console.WriteLine($"Exception: {ex.GetType()}. {ex.Message}");
                }
            }

            if (channel != null)
            {
                Console.WriteLine($"Closing session '{channel.SessionId}'.");
                await channel.CloseAsync();
            }
        }
```

balamcsd, 2017-11-27:
Thanks awcullen sir.
Does opening the channel & subscription has to be in continuous running while loop?

awcullen, 2017-11-27:
You don't have to use a while loop. In OpenOpcConnection, you can add an event handler like:
```
channel.Faulted += OnFaulted;

```
In OnFaulted, you reopen your connection and recreate your subscription.

balamcsd, 2017-11-27:
Thanks sir.
Also i am getting reactive error, in this scenerio and my windows service is stopped, 
probably because of this lines of code 
var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
normal try catch block is not catching this exception.
Can you please suggest how to handle it

sbradford02, 2018-02-09:
I'm having this same issue.  Did you guys get an answer for this?

awcullen, 2018-02-10:
Sorry for the delay.  A solution may be to provide an exception handler in the call to subscribe.

```
var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(
    pr =>
    {
        // loop thru all the data change notifications
        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
        foreach (var dcn in dcns)
        {
            foreach (var min in dcn.MonitoredItems)
            {
                Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
                count++;
            }
        }

    },
    ex =>
    {
        Console.WriteLine($"IObserver handled exception '{ex.GetType()}'. {ex.Message}");
    }
);

```

sbradford02, 2018-02-12:
That worked. Thanks!

sbradford02, 2018-02-12:
I now have an issue with the CreateMonitoredItemsAsync() call always timing out after I rerun my startup logic to recreate a new channel, re-register nodes, re-setup subscriptions, etc... after Kepware starts back up.  This is the exact same logic that works fine on startup of my service.  Here is what the debug log looks like for the channel once I rerun the startup sequence:

**Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Discovering endpoints of 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received OpenSecureChannelResponse, Handle: 1 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opened.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending GetEndpointsRequest, Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received GetEndpointsResponse, Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CloseSecureChannelRequest, Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closed.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Success discovering endpoints of 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Found certificate with subject alt name 'urn:NASLAP1267:OpcUAService'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentity: ''.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received OpenSecureChannelResponse, Handle: 1 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSessionRequest, Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSessionResponse, Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ActivateSessionRequest, Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ActivateSessionResponse, Handle: 3 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ReadRequest, Handle: 4
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ReadResponse, Handle: 4 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSubscriptionRequest, Handle: 5
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSubscriptionResponse, Handle: 5 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opened.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 6
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 7
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 8
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending RegisterNodesRequest, Handle: 9
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received RegisterNodesResponse, Handle: 9 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSubscriptionRequest, Handle: 10
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSubscriptionResponse, Handle: 10 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ReadRequest, Handle: 11
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received PublishResponse, Handle: 6 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 12
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ReadResponse, Handle: 11 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateMonitoredItemsRequest, Handle: 13**

I never get a response from the CreateMonitoredItemsRequest.  The exception from the CreateMonitoredItemsAsync() call is below:

**Workstation.ServiceModel.Ua.ServiceResultException: Timeout occurred while processing the request.
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1.ConfiguredTaskAwaiter.GetResult()
[... cut ...]
   at PSI.OPCUALib.HMIInterface.<ExecuteSubscribeToTagsAsync>d__65.MoveNext()**

I'm only trying to monitor one tag with the request in this example and I have verified its NodeID matches the initial NodeID (ns =99; i=4) it has on startup before I attempt this shutdown Kepware test.  I can always restart my whole service application and reconnect fine on startup but stopping Kepware and starting it back always results in the monitor request timing out.  Please let me know if you have any ideas.

Thanks

awcullen, 2018-02-13:
Thank you for the detailed info.  I can see that you have called ReqisterNodesAsync. That's a good idea if you plan to call ReadAsync, over and over.  For subscriptions, you don't need to do that. In fact, maybe that has confused the server a bit. You should try create the monitored item with the original nodeID.

sbradford02, 2018-02-13:
I removed the RegisterNodes call and started using the original string NodeIDs which works fine on startup.  However, on reconnection/recreation it is still getting stuck on the CreateMonitoredItemsRequest until it times out.  See below:

**Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Discovering endpoints of 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received OpenSecureChannelResponse, Handle: 1 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel opened.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending GetEndpointsRequest, Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received GetEndpointsResponse, Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closing.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CloseSecureChannelRequest, Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Channel closed.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Success discovering endpoints of 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opening.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Found certificate with subject alt name 'urn:NASLAP1267:OpcUAService'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: Opening session channel with endpoint 'opc.tcp://localhost:49320'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityPolicy: 'http://opcfoundation.org/UA/SecurityPolicy#Basic256'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: SecurityMode: 'SignAndEncrypt'.
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Information: UserIdentity: ''.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending OpenSecureChannelRequest, Handle: 1
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received OpenSecureChannelResponse, Handle: 1 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSessionRequest, Handle: 2
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Installed new security token 1.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSessionResponse, Handle: 2 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ActivateSessionRequest, Handle: 3
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ActivateSessionResponse, Handle: 3 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ReadRequest, Handle: 4
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ReadResponse, Handle: 4 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSubscriptionRequest, Handle: 5
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSubscriptionResponse, Handle: 5 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel: Trace: Channel opened.
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 6
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 7
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending PublishRequest, Handle: 8
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateSubscriptionRequest, Handle: 9
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received CreateSubscriptionResponse, Handle: 9 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending ReadRequest, Handle: 10
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Received ReadResponse, Handle: 10 Result: 0x00000000
Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel: Trace: Sending CreateMonitoredItemsRequest, Handle: 11
The thread '<No Name>' (0x9500) has exited with code 0 (0x0).
The thread '<No Name>' (0xb458) has exited with code 0 (0x0).
The thread '<No Name>' (0x468c) has exited with code 0 (0x0).
The thread '<No Name>' (0x953c) has exited with code 0 (0x0).
The thread '<No Name>' (0x7c64) has exited with code 0 (0x0).**

Any more ideas?

Thanks again

awcullen, 2018-02-14:
I would definitely test subscribing to the CurrentTime. Use NodeId.Parse("i=2258")

sbradford02, 2018-02-16:
Thanks for the reply.  As requested I tried that tag.  Doesn't seem to matter which tag I try.  If the channel has connected and then faults because Kepware has been shutdown, once Kepware is turned back on and after I create a new instance of the channel, open it, and subscribe, the **var response = await channel.CreateMonitoredItemsAsync(request);** always hangs and times out.  It is the exact same request that works fine with the initial channel.  The subscription request and everything else works fine with the new instance of the channel it's just the Create Monitored Items call that hangs.

sbradford02, 2018-02-16:
I finally figured this out.  My method that was wrapping the channel.CreateMonitoredItemsAsync() call was being called with a .Wait() which was causing a deadlock.  Thanks for your help.

awcullen, 2018-02-16:
Good detective work.

Andrew Cullen
914-374-5275

closed 2018-02-20 by awcullen