convertersystems/opc-ua-client #56: Problem getting version 2.1.1 into .NET 4.6.1 library

WoodsterDK, 2017-11-28:
Hi,
I have written a universal wrapper for interfacing with the opc-ua-client, but it does not seem to install correctly when adding the package from nuget.
I'm getting yellow warning at the System.Value.Tuble and the System.Net.Http references.

Any ideas of what to do?

awcullen, 2017-11-28:
You can check out the samples at https://github.com/convertersystems/opc-ua-samples. Many of the projects are .NET 4.6.1

WoodsterDK, 2017-11-28:
Hi Andrew, 
yes - already did - and I can see they reference the two assemblies correct, and without any warnings.

What I did was to create a new .NET Framework class library, and from Nuget add package "Workstation.UaClient" it looks like all the dependencies gets installed - but those two mentioned above does get a yellow warning mark.
I tried to force a reference to the dll in "packages\System.ValueTuple.4.4.0" but still a yellow warning.

Any ideas....?

awcullen, 2018-02-20:
I hope this was resolved. Please reopen if there is still an issue.

closed 2018-02-20 by awcullen