convertersystems/opc-ua-client #59: Decoding halted because of invalid data in the stream.

balamcsd, 2018-02-19:
Hai,
I am using Workstation.UaClient 2.0
when i try to connect to 
opc.tcp://My-PC:53530/OPCUA/SimulationServer
i am able to successfully connect to prosys OPC UA simulation server.
But when i try to create subscription  
In this line
BrowseResponse browseResponse = await channel.BrowseAsync(browseRequest);
it throw exception 
**Decoding halted because of invalid data in the stream.**
Same code works with kepserver.
Please help to fix this issue
Thanks

awcullen, 2018-02-19:
Thank you for reporting this issue.
I am using version 2.1.1 of Workstation.UaClient and version  2.2.2-109 of Prosys Simulation Server.
I have tested browsing to three levels using the following code and have not found the exception.
Please let me know if there was a particular NodeId you were browsing when the error occurred.

BR, 
Andrew
```
                Console.WriteLine("Step 4 - Browse the server namespace.");
                Console.WriteLine("+ Root");
                BrowseRequest browseRequest = new BrowseRequest
                {
                    NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = NodeId.Parse(ObjectIds.RootFolder), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
                };
                BrowseResponse browseResponse = await channel.BrowseAsync(browseRequest);
                foreach (var rd1 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
                {
                    Console.WriteLine("  + {0}: {1}, {2}", rd1.DisplayName, rd1.BrowseName, rd1.NodeClass);
                    browseRequest = new BrowseRequest
                    {
                        NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd1.NodeId, channel.NamespaceUris), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
                    };
                    browseResponse = await channel.BrowseAsync(browseRequest);
                    foreach (var rd2 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
                    {
                        Console.WriteLine("    + {0}: {1}, {2}", rd2.DisplayName, rd2.BrowseName, rd2.NodeClass);
                        browseRequest = new BrowseRequest
                        {
                            NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd2.NodeId, channel.NamespaceUris), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
                        };
                        browseResponse = await channel.BrowseAsync(browseRequest);
                        foreach (var rd3 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
                        {
                            Console.WriteLine("      + {0}: {1}, {2}", rd3.DisplayName, rd3.BrowseName, rd3.NodeClass);
                        }
                    }
                }
```

awcullen, 2018-02-20:
I successfully browsed all 2396 nodes using version 2.1.1 
Let us know if you have success too.

balamcsd, 2018-02-20:
What is Target framework?

awcullen, 2018-02-20:
.net framework 4.6.1

Andrew Cullen
914-374-5275

balamcsd, 2018-02-20:
Thanks.
I changed my framework from 4.5.2 -> 4.6.1
But when i try to execute 
await channel.OpenAsync();
It throws 
"Could not load file or assembly 'System.Net.Sockets, Version=4.1.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies. The system cannot find the file specified.":"System.Net.Sockets, Version=4.1.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a"}
Please suggest
Thanks

balamcsd, 2018-02-20:
Yes, after upgrading Workstation.UaClient 2.1.1 it is working in both prosys and kepware
Thanks

awcullen, 2018-02-20:
Fantastic!

closed 2018-02-20 by awcullen