convertersystems/opc-ua-client #60: Nuget fails on VS2017 .NET 4.5.1

cpipero, 2018-03-02:
Trying to start a console app in .NET 4.5.2, but VS errors on nuget saying:
<Error		Could not install package 'Workstation.UaClient 2.1.1'. You are trying to install this package into a project that targets '.NETFramework,Version=v4.5.2', but the package does not contain any assembly references or content files that are compatible with that framework. For more information, contact the package author.			0	>

Thanks

awcullen, 2018-03-02:
Please target .NET 4.6.1, which is supported by Windows 7, 8 and 10.

awcullen, 2018-03-02:
Or you can target .NET Core 2.0, which is supported on Windows 7sp1-10, MacOs and Linux. 
See: https://github.com/dotnet/core/blob/master/release-notes/2.0/2.0-supported-os.md