convertersystems/opc-ua-client #61: Windows service and ServiceResultException

andrea5586, 2018-03-08:
Hi everyone, 
i'm having a trouble i'm not able to figure out and I'm asking for your help.
I have a visual studio windows service project that uses this awesome package Workstation.UaClient

All works properly, i read/write tags on a OPC UA server on a siemens hmi TP1900 confort .

Sometimes it happens with no particular reason that the windows service crashes due to a ServiceResultException. 

On the EventViewer i can see the following stack trace:

```
Application: MyWinService.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: Workstation.ServiceModel.Ua.ServiceResultException
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel+<RequestAsync>d__115.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at Workstation.ServiceModel.Ua.SubscriptionServiceSet+<PublishAsync>d__3.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at Workstation.ServiceModel.Ua.ServiceExtensions+<WithCancellation>d__2`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(System.Threading.Tasks.Task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(System.Threading.Tasks.Task)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel+<PublishAsync>d__49.MoveNext()

Exception Info: System.AggregateException
   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(System.Exception)
   at System.Reactive.Stubs+<>c.<.cctor>b__2_1(System.Exception)
   at System.Reactive.AnonymousSafeObserver`1[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].OnError(System.Exception)
   at System.Reactive.Linq.ObservableImpl.Where`1+_[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].OnError(System.Exception)
   at System.Threading.Tasks.Dataflow.DataflowBlock+SourceObservable`1+ObserversState[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].NotifyObserversOfCompletion(System.Exception)
   at System.Threading.Tasks.Dataflow.Internal.Common+<>c.<ThrowAsync>b__20_0(System.Object)
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context(System.Object)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback()
```

The strange thing is that i get the exception also if my code is on try catch block.

Does anyone else has this problem?

awcullen, 2018-03-09:
I believe you can catch this exception by adding a second action to the Subscribe method.
pr => { }, ex => { }

Andrew Cullen
914-374-5275

andrea5586, 2018-03-09:
Thank you, I'll give it a try and let you know!

andrea5586, 2018-03-10:
I tried as you suggested and the windows service now doesn't crash.
I would like to understand which is the problem: i added the following lines
```
pr => {...}
, 
ex => 
{
    StringBuilder sb = new StringBuilder();
    sb.AppendLine("ERROR ON Subscribe:");
    sb.AppendLine($"\t{ex.Message}");
    sb.AppendLine($"\t{ex.StackTrace}");
    Log(sb.ToString());
}
```
But the only thing i get in the log is:
```
ERROR ON Subscribe:
	One or more errors occurred.
	
```
No stack trace. Is there a way to get it and better understand which are these errors?
Thanks

awcullen, 2018-03-10:
Looks like an AggregateException. Try
```
    var ex1 = ex.GetBaseException();
    StringBuilder sb = new StringBuilder();
    sb.AppendLine("ERROR ON Subscribe:");
    sb.AppendLine($"\t{ex1.Message}");
    sb.AppendLine($"\t{ex1.StackTrace}");
    Log(sb.ToString());
  ```

andrea5586, 2018-03-12:
Thank you for the advice. 
I'll close the issue since the first solution permitted me to catch the exception. 
Still haven't had any other exceptions in days to test for AggregateException. 
Thanks a lot for your help!

closed 2018-03-12 by andrea5586