convertersystems/opc-ua-client #63: Any planing with the framework for the future?

OmidID, 2018-07-16:
I can see the last push for 10 months ago it is.
Do you have any plan for update and release it for other platforms ?

awcullen, 2018-07-16:
Hello,
Indeed, it is a good time for an update.  The Opc UA specification has been updated to v1.04 with many new features.  New session-less services,  improved methods for describing a custom data structure, and PubSub.  What new features or platforms would be useful to you?

OmidID, 2018-07-16:
Hello,
As you know .NetFx will be replace with .Net Core so i suggest to support .Net Standard Library.
For the moment we are using Mono + Linux-ARM and we thinking to use your library as client.
And we will upgrade to .Net Core for next generation.
By the way we are using self-signed certificate and it's would be perfect to support custom certificate authorization.
Thanks.

closed 2019-07-28 by awcullen