convertersystems/opc-ua-client #65: Unhandled exception when connection lost

basprins, 2018-10-22:
Dear Andrew,
I am trying to make our application robust against connection lost with the PLC (thus opcua connection lost). 

There are two moments I actively wait for a response over the opcua channel: reading and writing. The publishing I have no control over, as it's an event based mechanism triggered in your library.

So I thought I could simply try/catch the read/write calls:

     var writeResult = await session.WriteAsync(...)
     var writeResult = await session.ReadAsync(...)

When I restart the PLC (thus the connection is lost) I get a ServiceResultException. I catch that exception, and handle it, but immediately after my AppDomainUnhandledExceptionHandler fires: 

2018-10-22 20:42:48.657 [ERROR][thread: 3][Startup] Unhandled exception occurred
System.AggregateException: One or more errors occurred. ---> Workstation.ServiceModel.Ua.ServiceResultException: Timeout occurred while processing the request.
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSecureChannel.<RequestAsync>d__115.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.SubscriptionServiceSet.<PublishAsync>d__3.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.ServiceExtensions.<WithCancellation>d__2`1.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Workstation.ServiceModel.Ua.Channels.UaTcpSessionChannel.<PublishAsync>d__49.MoveNext()
   --- End of inner exception stack trace ---

Is this something you recognize? Is it something I can fix, or is this by design in your library?

awcullen, 2018-10-22:
You may be seeing the exception in the publishResponse handler.  There's an extra parameter where you can add an exception handler.
For example:

```
var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(
    pr =>
    {
        // loop thru all the data change notifications
        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
        foreach (var dcn in dcns)
        {
            foreach (var min in dcn.MonitoredItems)
            {
                Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
            }
        }
    },
    ex =>
    {
        Console.WriteLine("Exception in publish response handler: {0}", ex.GetBaseException().Message);
    });

```

basprins, 2018-10-23:
Brilliant. Thx!!

closed 2018-10-23 by basprins