convertersystems/opc-ua-client #69: Using Subscriptionbase in and console application

mlebjerg, 2018-12-06:
Hi I want to use this framework for MVC, and have tried to modifying your StatusHMI to work in MVC.
I have seen in another issue ( https://github.com/convertersystems/opc-ua-client/issues/50 ) where you said the attributes have to be observed, and it does work in a forms project, but i cannot for the life of me get the variables, is there anyway to observe it from let's say a console application?

awcullen, 2018-12-06:
Hi,  I prefer easy as well..  Would you mind sharing an example MVC application and indicate where you might want the Subscription to be created?

Andrew

mlebjerg, 2018-12-06:
Wow thanks for the quick answer!
I made a console application from what @udoliess posted in #50, and it works, now the question is just how to write to it :) 

The idea is to have a few UaApplications if there are more machines, and be able to read and write to certain attributes.
For example: The user clicks the start button, that makes an ajax call to the controller where there is a method that changes the value of "ns=6;s=::Program:Cube.Command.CntrlCmd" to 2.

How do i write to it?

I have tried this:

`using System;
using System.ComponentModel;
using System.Net;
using System.Threading;
using Workstation.ServiceModel.Ua;
[Subscription(endpointUrl: "opc.tcp://localhost:4840", publishingInterval: 50)]
class Program : SubscriptionBase
{
    static void Main(string[] args)
    {
        using (var app = new UaApplicationBuilder()
            .SetApplicationUri($"urn:{Dns.GetHostName()}:Test")
            .AddMappedEndpoint("opc.tcp://localhost:4840", "opc.tcp://localhost:4840", SecurityPolicyUris.None)
            .Build())
        {
            app.Run();
            var prog = new Program();
            PropertyChangedEventHandler dummy = delegate { };
            try
            {
                prog.PropertyChanged += dummy; // needed to start background activities
                ConsoleKeyInfo key = new ConsoleKeyInfo();

                while (true)
                {
                    key = Console.ReadKey(true);
                    switch (key.Key)
                    {
                        case ConsoleKey.D1:
                            prog.programCubeCommandCntrlCmd = 1;
                            break;
                        case ConsoleKey.D2:
                            prog.programCubeCommandCntrlCmd = 2;
                            break;
                        case ConsoleKey.D3:
                            prog.programCubeCommandCntrlCmd = 3;
                            break;
                        case ConsoleKey.D4:
                            prog.programCubeCommandCntrlCmd = 4;
                            break;
                        case ConsoleKey.D:
                            prog.programCubeCommandCntrlCmd = 5;
                            break;
                    }
                }
            }
            finally
            {
                prog.PropertyChanged -= dummy;
            }
        }
    }

    /// <summary>
    /// Gets or sets the value of ProgramCubeCommandCntrlCmd.
    /// </summary>
    [MonitoredItem(nodeId: "ns=6;s=::Program:Cube.Command.CntrlCmd")]
        public Int32 ProgramCubeCommandCntrlCmd
        {
            get { return this.programCubeCommandCntrlCmd; }
            set { this.SetProperty(ref this.programCubeCommandCntrlCmd, value); }
        }
        private Int32 programCubeCommandCntrlCmd;

}`

mlebjerg, 2018-12-06:
You have made really great framework! It is so easy that i can't believe it! 
I can fork your samples and add MVC samples if you are interested.

awcullen, 2018-12-06:
That would be fantastic!

tejasklodha, 2018-12-06:
Yes,It's really wonderful code.... And after year still I m not able to
find code like this...

Tejas Lodha

mlebjerg, 2018-12-06:
I looked at a lot of these framework, they are either good, and crazy expensive, or complicated, no Docs and free.

I will see what i can do! In regards to MVC samples :) 
Closing this one thanks for the help!

closed 2018-12-06 by mlebjerg