convertersystems/opc-ua-client #70: Array change event

tejasklodha, 2018-12-25:


tejasklodha, 2018-12-25:
Dear Sir,

I have below subscription. 

`[Subscription(endpointUrl: "opc.tcp://143.62.215.156:4840", publishingInterval: 500, keepAliveCount: 20)]`

As my OPC machine display, Value change frequently but my array is not changed frequently and get data in every 5 to 10 second my actual data is changed within 1 second.

Can you help If I am doing something wrong?
My array definition is as below

`[MonitoredItem(nodeId: "ns=6;s=::AsGlobalPV:OPCProcParaValue")]

        public UInt32[] Global_PVOPCProcParaValue

        {
            get { return this.global_PVOPCProcParaValue; }
            private set
            {
                this.SetProperty(ref this.global_PVOPCProcParaValue, value);
                if (global_PVOPCProcParaValue != null)

                {
                    CallingMethod(global_PVOPCProcParaValue);
                }

            }

        }`

awcullen, 2018-12-31:
Hi,
I tried connecting to UnifiedAutomation's UaCPPServer with a subscription like: 
```
    [Subscription(endpointUrl: "opc.tcp://localhost:48010", publishingInterval: 500, keepAliveCount: 20)]
    public class MainViewModel : SubscriptionBase
    {
        [MonitoredItem(nodeId: "ns=2;s=Demo.Dynamic.Arrays.UInt32")]
        public UInt32[] TestArray
        {
            get { return this.testArray; }
            private set { this.SetProperty(ref this.testArray, value); }
        }

        private UInt32[] testArray;
    }
```

TestArray updated at 500 ms as requested.  Your code looks good as well.

tejasklodha, 2019-01-25:
Still, my array event gets data after  2 seconds even I can able to see live data in millisecond.

can you give the example that gets continuous data in every 500 milliseconds even aray chnage or not?

awcullen, 2019-01-25:
to get updates whether data changes or not:
```
        [MonitoredItem(nodeId: "ns=2;s=Demo.Dynamic.Arrays.UInt32", dataChangeTrigger: DataChangeTrigger.StatusValueTimestamp)]
        public UInt32[] TestArray
        {
            get { return this.testArray; }
            private set { this.SetProperty(ref this.testArray, value); }
        }

        private UInt32[] testArray;
```

tejasklodha, 2019-01-25:
And How can we define timestamp?
Like, need data in every 500 Mili second then what needs to be changed in code?

tejasklodha, 2019-01-25:
Dear Friend,

I tried below code but not working for static value.

[MonitoredItem(nodeId: "ns=2;s=Demo.Static.Arrays.Enumeration", dataChangeTrigger: DataChangeTrigger.StatusValueTimestamp)]
        public Int32[] ArraysEnumeration
        {
            get { return this.arraysEnumeration; }
            set { this.SetProperty(ref this.arraysEnumeration, value); }
        }
        private Int32[] arraysEnumeration;

Need your help....

awcullen, 2019-01-25:
DataChangeTrigger.StatusValueTimestamp  means "publish each sample, whether the values in the array have changed or have not changed". 

I suggest you begin with tool such as UaExpert to view "ns=6;s=::AsGlobalPV:OPCProcParaValue" and understand how often it changes.

awcullen, 2019-01-25:
You bring up a good point about "ns=2;s=Demo.Static.Arrays.Enumeration". You are correct that StatusValueTimestamp does not cause the demo server to send a stream of values (changed or unchanged).  This is because the option seems to only work for servers that poll for data.

tejasklodha, 2019-01-25:
Ok..

In testing, I am using UA demo server as per below image.
[image]

And As per I understand that I can not check this `DataChangeTrigger.StatusValueTimestamp` functionality with this UA demo server.
If I want to check this functionality then I must use Live server like siemens or B&R PLC.
Am I correctly understand?

awcullen, 2019-01-27:
To understand the performance of the server,  use a tool such as UaExpert to subscribe to "ns=6;s=::AsGlobalPV:OPCProcParaValue" and verify how often you receive data changes.  You can change the Trigger value of the and see if there is any difference. 
[image]

abrasat, 2019-02-01:
Is it possible to set for a monitored item in the OPC-UA client something like "On value change (but not faster than 50 msec), and every 5 seconds if value does not change"

awcullen, 2019-02-01:
Not easily.  You can make two monitored items. First with fast sample rate and second with slow rate and  DataChangeTrigger.StatusValueTimestamp.  

Did you find out the cause of your original question "Array change event is slow" ?

closed 2019-07-28 by awcullen