convertersystems/opc-ua-client #71: Can this be used to connect to OPC DA (Classic) server?

RaymondGemalto, 2019-02-28:
Are OPC UA backward compatible with OPC DA; I.e. The Class OPC DA server?

awcullen, 2019-02-28:
You would need an OPC UA Server in the middle. The OPC Foundation provides COM UA Wrapper, an OPC UA Server that wraps an OPC DA Server and with that enables an OPC UA Client to access information from the DA Server. 
Further, there are commercial solutions that provide a gateway (wrapper and proxy together) such as https://www.unified-automation.com/products/wrapper-and-proxy/uagateway.html.  
Another solution also includes a tunneler, see https://www.matrikonopc.com/opc-ua/products/opc-ua-tunneller.aspx

RaymondGemalto, 2019-03-01:
Actually, my requirement is very simple; access OPC DA (classic) server and provide these data to somewhere else. Don't need OPC UA nor OPC UA Server. I read OPC UA Client can be used to read OPC DA (Classic) data too. But I can't find the right document without paying for commercial SDK.

That's why I'm hoping this project can provide me with that tool to access OPC DA directly.

awcullen, 2019-03-01:
OPC UA clients can not read DA servers directly.  There must be a wrapper/gateway in between.

closed 2019-03-01 by awcullen