convertersystems/opc-ua-client #72: Private local key is null

basprins, 2019-05-09:
Hi Andrew

Also tried to connect using the UaApplicationBuilder, but I am having trouble getting it to work.

I altered the example on your git homepage slightly to my needs

App.xaml.cs

```
        private async void App_OnStartup(object sender, StartupEventArgs e)
        {
            _uaApplication = new UaApplicationBuilder()
                .SetApplicationUri($"urn:{Dns.GetHostName()}:Workstation.StatusHmi")
                //.SetIdentity(async (args) => await Task.FromResult(new AnonymousIdentity()))
                .AddMappedEndpoint("opc.tcp://localhost:4840", new EndpointDescription
                {
                    EndpointUrl = "opc.tcp://localhost:4840",
                    SecurityMode = MessageSecurityMode.None,
                })
                .Build();

            _uaApplication.Run();
            
            var window = new MainWindow
            {
                DataContext = new MainViewModel()
            };
            window.Show();
        }
```

Viewmodel:

```
    [Subscription(endpointUrl: "opc.tcp://localhost:4840", publishingInterval: 500, keepAliveCount: 20)]
    public class MainViewModel : SubscriptionBase
    {
        private double _lsb;
        public RelayCommand ConnectCommand { get; }
        public RelayCommand DisconnectCommand { get; }

        [MonitoredItem(nodeId: "i=6")]
        public double Lsb
        {
            get => _lsb;
            set => SetProperty(ref _lsb, value);
        }
    }
```

The opcua property details via uaexpert:

[image]

When I run the above app, there is no connection created. I can try to set a value, but it will throw an exception that no client is connected. 

I also tried to get the actual channel that should have been created:

```
            _uaApplication = new UaApplicationBuilder()
                .SetApplicationUri($"urn:{Dns.GetHostName()}:Workstation.StatusHmi")
                //.SetIdentity(async (args) => await Task.FromResult(new AnonymousIdentity()))
                .AddMappedEndpoint("opc.tcp://localhost:4840", new EndpointDescription
                {
                    EndpointUrl = "opc.tcp://localhost:4840",
                    SecurityMode = MessageSecurityMode.None,
                })
                .Build();

            _uaApplication.Run();
            var channel = await _uaApplication.GetChannelAsync("opc.tcp://localhost:4840");
```

But that throws the exception "Local private key is null".

What am I doing wrong?

awcullen, 2019-05-11:
You were so close!  Sorry to leave you guessing.  This seems to work for me.
```
uaApplication = new UaApplicationBuilder()
                .SetApplicationUri($"urn:{Dns.GetHostName()}:Workstation.StatusHmi")
                //.SetIdentity(async (args) => await Task.FromResult(new AnonymousIdentity()))
                .AddMappedEndpoint("opc.tcp://localhost:4840", new EndpointDescription
                {
                    EndpointUrl = "opc.tcp://localhost:4840",
                    SecurityPolicyUri = SecurityPolicyUris.None,
                })
                .Build();

```

basprins, 2019-05-13:
Ah that works! Thx!

closed 2019-05-13 by basprins