convertersystems/opc-ua-client #73: Unable to update server with SetProperty

basprins, 2019-05-13:
Hi Andrew,

I seem to have an issue updating an monitored item from client side. I have a working subscription now, and data changed in the server (BnR) is updated in the client as expected. But when I press a button which just sets the property, nothing happens in the server (and the TextBlock in my C# client is shown with a red border).

My Subscription:
```

    [Subscription(endpointUrl: "opc.tcp://localhost:4840", publishingInterval: 500, keepAliveCount: 20)]
    public class MainViewModel : SubscriptionBase
    {
        private uint _rcwcPort;
        public RelayCommand Bla { get; }

        [MonitoredItem(nodeId: "ns=6;s=::AsGlobalPV:FlowAndTempature.Config.RcwcTcpPort")]
        public uint RcwcPort
        {
            get => _rcwcPort;
            set => SetProperty(ref _rcwcPort, value);
        }

        public MainViewModel()
        {
            Bla = new RelayCommand(Connect);
        }

        private void Connect()
        {
            RcwcPort = 1;
        }
    }
```

The UaApplicationBuilder as reference:

```
        private async void App_OnStartup(object sender, StartupEventArgs e)
        {
            _uaApplication = new UaApplicationBuilder()
                .SetApplicationUri($"urn:{Dns.GetHostName()}:BR/UA/EmbeddedServer")
                .SetIdentity(async args => await Task.FromResult(new AnonymousIdentity()))
                .AddMappedEndpoint("opc.tcp://localhost:4840", new EndpointDescription
                {
                    EndpointUrl = "opc.tcp://localhost:4840",
                    SecurityMode = MessageSecurityMode.None,
                    SecurityPolicyUri = SecurityPolicyUris.None,
                })
                .Build();

            _uaApplication.Run();
            var window = new MainWindow
            {
                DataContext = new MainViewModel()
            };
            window.Show();
        }
```

When I start my PLC simulation, rcwcport is set to 0.
I can change the value in the PLC (to 4 e.g.), the change is picked up in the C# client:

[image]

[image]

When I press the button in my C# app, the view INotifyChanged property in the viewmodel works fine, as the view shows the new value for RcwcPort (which is hardcoded to 1). But it's validation fails as it gets a red border:

[image]

The server is not updated, as its value remains 4.

[image]

And of course the same in automation studio :
[image]

awcullen, 2019-05-14:
My first guess is that the data type does not match exactly.  I've noticed that many servers are very picky about the data type when writing.  So instead of uint (which is 32bit in c# I believe) try uint16.
```
    private uint16 _rcwcPort;

    [MonitoredItem(nodeId: "ns=6;s=::AsGlobalPV:FlowAndTempature.Config.RcwcTcpPort")]
    public uint16 RcwcPort

basprins, 2019-05-15:
Hi Andrew, 
Bullseye. It works when the type is a ushort. The weird thing is, I also tried a bool, which is also rejected at some point. That surprised me, because I'd expect that a bool would be a bool on both sides, right? In UaExpert the property is shown as property with a "FalseState"  and "TrueState". That would tell me that it's a boolean all the way, not that it's boxed to an integer or something like that (0/1).

In any event, the fact that it's so sensitive scares me a little bit. I can imagine that my colleagues might struggle understanding why something isn't working.

I think my "overall question" is more about which approach should I take with your library. The rough requirements:
- communicate between C# and PLC server (BnR)
- log traces of variables in C# (so I need to have the date/time for samples of the specific monitored item

a) Should I go for the Attributed approach with [Subscription(..)] and [MonitoredItem(...)], or
b) Should I go for instantiating a UaTcpSessionChannel myself, and write a bit of user code on top of your library in order to "read/write/receive updates" for opcua properties in my C# application.

Is a) intended as a really quick way to use opcua, with a some restrictions? 
In other words, is b) the way to go considering the requirements for making trace logs of the samples + timestamps.

PS: Would this be useful information on your github wiki page ? It's rather empty at the moment :). 

I hope you can make sense of what I am asking, and looking forward to your honest opinion (if it's a stupid approach, by all means tell me rather sooner than later! :)) of what is the best approach for my case.

Some code as reference for what I wrote above below.

Thanks a lot in advance!

User code to create a 'OpcuaProperty' which can be used to read/write/monitor:

`            _initializeCommandProperty = opcua.AddOpcuaProperty<bool>("Some boolean in the server");
`

Inside my own 'OpcuaClient', I just store the property in my own List (registered properties). 
```
        public IOpcuaProperty<T> AddOpcuaProperty<T>(
            string identifier,
            Expression<Func<T>> propertyExpression = null,
            bool isMonitoredItem = true,
            bool logChanges = true)
        {
            var opcuaProperty = new OpcuaProperty<T>(identifier, _uniqueClientHandle++, propertyExpression, isMonitoredItem);
            _registeredOpcuaProperties.Add(opcuaProperty);
            return opcuaProperty;
        }
```

When the C# app wants to connect to the server, the 'OpcuaClient' also subscribes the 'OpcuaProperty's that are marked as monitored item:

```
        private async void Subscribe()
        {
            var subscriptionRequest = new CreateSubscriptionRequest
            {
                /*
                 * Specifies the time between publish requests from the server. If it's set to 1000, the server will publish updates on monitored items
                 * every second. 
                 */
                RequestedPublishingInterval = 100,
                /*
                 * The server will send publish requests to check if the connection is still OK. It will do that by multiplying the publish interval 
                 * with the max keep alive count. Thus, if the publish interval is set to 1000, and the keep alive count is set to 10, the server
                 * will send an empty publish request to validate the connection every 10 seconds. 
                 */
                RequestedMaxKeepAliveCount = 10,
                RequestedLifetimeCount = 1,
                PublishingEnabled = true
            };

            var subscriptionResponse = await _channel.CreateSubscriptionAsync(subscriptionRequest);
            _channel.Faulted += (sender, args) =>
            {
                // todo figure out what to do on this event
            };
            _channel.Closing += (sender, args) =>
[... 98 lines cut ...]
                var errorMessage = StatusCodes.GetDefaultMessage(writeResult.Results[0].Value);
                throw new OpcuaCommunicationException($"Unable to write property '{NodeId}': {errorMessage}");
            }
        }

        public void Update(DataValue value)
        {
            if (value.Value != null)
            {
                _value = (T) value.Value;

                if (_propertyExpression != null)
                {
                    var memberExpression = (MemberExpression) _propertyExpression.Body;
                    var propertyInfo = (PropertyInfo) memberExpression.Member;
                    var propertyOwnerExpression = (ConstantExpression) memberExpression.Expression;
                    var propertyOwner = Expression.Lambda(propertyOwnerExpression).Compile().DynamicInvoke();

                    propertyInfo.SetValue(propertyOwner, _value, null);
                }

                ValueChanged?.Invoke(this, new ValueChangedEventArgs<T>(_value));
            }
        }
```

awcullen, 2019-05-15:
I use the attributes approach when I am building an HMI app based on WPF, UWP, or Xamarin.  These platforms support the MVVM (Model, View, View Model) pattern. And SubscriptionBase makes a good view model.
I use the UaTcpSessionChannel directly everywhere else. 

Also you can use UaExpert to find the exact type of value a variable is expecting. It is documented in the property 'DataType'.

Your code looks great.  Share it on GitHub when you can.

basprins, 2019-05-15:
Well, we are using WPF with MVVM here, so maybe I should give the attributes approach another try. I do feel more "in control" with the UaTcpSessionChannel though. I can also write code to create the identifiers for the monitored items (which would have to be constant using attributes). But I guess that's just personal preference.

Regarding the DataType in uaexpert: that's helpful information too. The boolean is exposed as :
- IdentifierType = "Numeric"
- Identifier = 1 [Boolean]

Thanks for the answer, helps me get more confident in moving forward. If I get to it I'll drop the code in a github repo and post the project here (you can do whatever you like with it).

:+1: thx a lot!