convertersystems/opc-ua-client #74: Subscription stops working without notification?

basprins, 2019-05-13:
Hi Andrew,
Sorry I have to bother you so much... 

I've created an subscription and try to understand how it behaves (so that I can make my app robust for things that can occur).

I put a breakpoint in the data changed handler intentionally so that I can see how the subscription responds to errors. If I keep pressing F5 (continue) it always keeps hitting the breakpoint as I expect. But, when I stop pressing F5 for a few seconds, the behavior is different.

1) When I block all threads due to a breakpoint for a long time (say a minute), the exception handler attached to the subscription gives an invalid subscription exception. That makes sense to me, I can react on that.

2) When I block all threads due to a breakpoint for a shorter time, say 10 seconds, then pressing F5 will give me 2 or 3 more publish requests, and afterwards it simply stops. No exception is notified, no event handlers like faulting/closing/closed are triggered. It just plain stops. 

Here the code where I create the subscription :

```
            var subscriptionRequest = new CreateSubscriptionRequest
            {
                /*
                 * Specifies the time between publish requests from the server. If it's set to 1000, the server will publish updates on monitored items
                 * every second. 
                 */
                RequestedPublishingInterval = 100,
                /*
                 * The server will send publish requests to check if the connection is still OK. It will do that by multiplying the publish interval 
                 * with the max keep alive count. Thus, if the publish interval is set to 1000, and the keep alive count is set to 10, the server
                 * will send an empty publish request to validate the connection every 10 seconds. 
                 */
                RequestedMaxKeepAliveCount = 10,
                RequestedLifetimeCount = 1,
                PublishingEnabled = true
            };

            var subscriptionResponse = await _channel.CreateSubscriptionAsync(subscriptionRequest);
            _channel.Faulted += (sender, args) => { };
            _channel.Closing += (sender, args) =>
            {

            };
            _channel.Closed += (sender, args) =>
            {
                ;
            };

            var id = subscriptionResponse.SubscriptionId;

            var itemsToCreate = new[]
            {
                new MonitoredItemCreateRequest
                {
                    ItemToMonitor = new ReadValueId
                    {
                        NodeId = NodeId.Parse("ns=6;s=::AsGlobalPV:FlowAndTempature.Config.RcwcTcpPort"),
                        AttributeId = AttributeIds.Value
                    },
                    MonitoringMode = MonitoringMode.Reporting,
                    RequestedParameters = new MonitoringParameters
                    {
                        ClientHandle = 12345,
                        SamplingInterval = 1000,
                        QueueSize = 10,
                        DiscardOldest = true
                    }
                }
            };
            var itemsRequest = new CreateMonitoredItemsRequest
            {
                SubscriptionId = id,
                ItemsToCreate = itemsToCreate,
            };
            var itemsResponse = await _channel.CreateMonitoredItemsAsync(itemsRequest);

            var stopwatch = Stopwatch.StartNew();
            var token = _channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
            {
                // loop thru all the data change notifications
                var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
                foreach (var dataChangeNotification in dcns)
                {
                    foreach (var monitoredItemNotification in dataChangeNotification.MonitoredItems)
                    {
                        Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {monitoredItemNotification.ClientHandle}; value: {monitoredItemNotification.Value}");
                    }
                }
                Debug.WriteLine($@"now: {stopwatch.ElapsedMilliseconds}");
                stopwatch.Restart();
            }, exc =>
            {
                
            }, () =>
            {
                
            });
```

Can you explain why it does what it does, and how I can identify that subscription stopped working in a robust fashion?

awcullen, 2019-05-15:
You may want to log the exceptions, like:
```
            var loggerFactory = new LoggerFactory();
            loggerFactory.AddDebug(LogLevel.Trace);

            var channel = new UaTcpSessionChannel(
                appDescription,
                certificateStore,
                ShowSignInDialog,
                discoveryUrl,
                loggerFactory: loggerFactory);
```

basprins, 2019-05-16:
How do I configure the logger to write to a file (and tell which file it should log to)? 

PS: I don't have the LoggerFactory.AddDebug(LogLevel) method on the loggerFactory.

awcullen, 2019-05-16:
Please add Nuget package Microsoft.Extensions.Logging.Debug 

This package traces info to the Debug Window.

awcullen, 2019-05-16:
To write a log file, You can install Nuget package Serilog.Extensions.Logging.File
The instructions are here: https://github.com/serilog/serilog-extensions-logging-file

basprins, 2019-05-16:
Ah excellent! Thanks for pointing that out. I can't try this before
next week, so this will have to wait a little longer.

awcullen, 2019-07-04:
May I close this issue?

basprins, 2019-07-04:
Hi Andrew, haven't looked at the logging part, nor did I investigate this issue further. When it becomes an actual issue I will get back to you. Thx for all the support!

closed 2019-07-04 by basprins