convertersystems/opc-ua-client #77: Bad_WriteNotSupported error from KepwareOPC Server

yda86, 2019-05-24:
When I try write to "tag" on Kepware OPC I get that error.Code below:
`        static async void TestOpc()
        {
            loggerFactory = new LoggerFactory();
            loggerFactory.CreateLogger("logger");

            localDescription = new ApplicationDescription
            {
                ApplicationName = "Workstation.UaClient",
                ApplicationUri = $"urn:{Dns.GetHostName()}:Workstation.UaClient",
                ApplicationType = ApplicationType.Client
            };

            certificateStore = new DirectoryStore("F:\\OPC\\opc-ua-client-master\\UaClient");
            var userCert = default(X509Certificate);
            //var userKey = default(RsaKeyParameters);
            var certParser = new X509CertificateParser();
            var userCertInfo = new FileInfo("F:\\OPC\\opc-ua-client-master\\UaClient\\Key.snk");
            if (userCertInfo.Exists)
            {
                using (var crtStream = userCertInfo.OpenRead())
                {
                    var c = certParser.ReadCertificate(crtStream);
                    if (c != null)
                    {
                        userCert = c;
                    }
                }
            }
            else
            {
                Console.Write("cert not exist");
            }
            /*var userKeyInfo = new FileInfo(Path.Combine(pkiPath, "user", "private", "ctt_usrT.pem"));
            if (userKeyInfo.Exists)
            {
                using (var keyStream = new StreamReader(userKeyInfo.OpenRead()))
                {
                    var keyReader = new PemReader(keyStream);
                    var keyPair = keyReader.ReadObject() as AsymmetricCipherKeyPair;
                    if (keyPair != null)
                    {
                        userKey = keyPair.Private as RsaKeyParameters;
                    }
                }
            }*/
            if (userCert != null)
            {
                System.Security.Cryptography.X509Certificates.X509Certificate2 certificate = new
                    System.Security.Cryptography.X509Certificates.X509Certificate2();
                certificate.Import(userCert.GetEncoded());
                x509Identity = new X509Identity(userCert, null);
            }

            var channel = new UaTcpSessionChannel(
                localDescription,
                certificateStore,
                new AnonymousIdentity(),
                EndpointUrl,
                loggerFactory: loggerFactory);
            try
            {
                await channel.OpenAsync();
            }
            catch (Exception ex)
            {
                Console.WriteLine(ex.Message);
            }
            
            var writeRequest = new WriteRequest
            {
                NodesToWrite = new[] {
                    new WriteValue {
                        NodeId = NodeId.Parse("ns=2;s=Sadapter.SAP_ADAPTER.N_BAK"),
                        AttributeId = AttributeIds.Value,
                        IndexRange = "0:200",
                        Value = new DataValue("hello world")
                        /*Value = new DataValue(new Variant("hello world!!!"),
                        0,
                        DateTime.MinValue,
                        0,
                        DateTime.MinValue,
                        0
                        )*/
                    }
                }
            };
            var writeResult = await channel.WriteAsync(writeRequest);
            Console.WriteLine("{0} ", writeResult.Results[0].ToString() );

            var readRequest = new ReadRequest();
            readRequest.NodesToRead =
                new[] { new ReadValueId { NodeId = NodeId.Parse("ns=2;s=Sadapter.SAP_ADAPTER.N_BAK"), AttributeId = AttributeIds.Value } };
            var readResult = await channel.ReadAsync(readRequest);
            DataValue dvR = readResult.Results[0];
            Console.WriteLine("{0} ", dvR.Variant.Value);
            Console.Read();
            await channel.CloseAsync();           
        }`
I do someone wrong ? Or it library problem ?

awcullen, 2019-05-24:
Hi yda86,
I don't see the error code. So I am guessing that the Write is having trouble with:
 IndexRange = "0:200"
What happens if you remove that?
Andrew

yda86, 2019-05-25:
> Hi yda86,
> I don't see the error code. So I am guessing that the Write is having trouble with:
> IndexRange = "0:200"
> What happens if you remove that?
> Andrew

Yes You right, problem really with IndexArrange.Thank You!

closed 2019-05-25 by yda86