convertersystems/opc-ua-client #83: The user identity token is not valid

mcsreddy17, 2019-06-17:
Hi, thank you for all your work. Currently, I'm trying to test out username and password support by creating my own opc client, and I'm seeing the following error when calling 'await channel.OpenAsync()' on the client.

Error: [0x80200000]="The user identity token is not valid.

awcullen, 2019-07-04:
I would test connecting with this username and password using a tool such as OPCExpert by Unified Automation.

mcsreddy17, 2019-07-05:
It is working with OPCUAExpert tool. In fact,

awcullen, 2019-07-05:
Okay.  Could you test if this occurs when using SecurityPolicy None and when using Basic256Sha256?

awcullen, 2019-07-28:
If this is still an problem, please reopen issue.

ricardosousa1231, 2020-10-02:
same problem, it still occurs with both security policies

awcullen, 2020-10-03:
Hi, 
Would you share some more details of the problem?  What is the server your connecting to? What is the SecurityPolicy? What error are you receiving?
Today I test the software with UnifiedAutomation's UaCPPServer, using username/password authentication  with all five security policies.  All is connecting correctly.

Thank you

awcullen, 2020-10-03:
started new issue #168

closed 2020-10-03 by awcullen

ricardosousa1231, 2020-10-06:
well , I am connecting through a VPN to a client's server. If I host a server on my computer it works too for me. I don't know if the problem is the server or the connection ...

krstev89, 2024-03-26:
> Hi, thank you for all your work. Currently, I'm trying to test out username and password support by creating my own opc client, and I'm seeing the following error when calling 'await channel.OpenAsync()' on the client.
> 
> Error: [0x80200000]="The user identity token is not valid.

did you fix the issue. I have the same problem and can not understand what is causing it.

ans-mindsandco, 2026-08-31:
I also have the problem. I can connect with UA Expert, but if i use prosys browser client, then i get the Bad_IdentityTokenInvalid