convertersystems/opc-ua-client #84: System.IO.FileLoadException : Could not load file or assembly 'System.Threading.Tasks.Dataflow, Version=4.6.2.0

basprins, 2019-06-26:
Hi Andrew,
I am struggling for days to get my build working. I am using uaclient 2.2.2, in an environment where we don't connect to nuget.org directly. UaClient nuget package and its dependencies are added to our nuget repository, with one exception:

System.Threading.Tasks.DataFlow 4.9.0 is available instead of 4.8.0 (the minimum required version for UaClient).

I can build the software, but when I run my unit tests, it fails on a FileLoadException :
System.IO.FileLoadException : Could not load file or assembly 'System.Threading.Tasks.Dataflow, Version=4.6.2.0

The exception gets thrown when I **retrieve the Object of the mocked interface**. This interface has a method with a method which has the UaTcpSessionChannel as argument. UaTcpSessionChannel uses classes from the System.Threading.Tasks.Dataflow namespace.

I am able to reproduce the exact problem in a small solution, which I uploaded to github:

https://github.com/basprins/uaclientdataflow

If you run the unit test in "'ClassLibrary2" you can see the exception yourself. How can I resolve this? 

Thx!

awcullen, 2019-07-02:
I found I had to uncheck the nuget package sources from Nexus.

[image]

basprins, 2019-07-02:
Hi Andrew,
Pretty sure that's not the solution. I just pushed a commit to my github repo, now it only uses nuget.org.
If you delete the [solution]\Packages folder and rebuild the solution, you will have only dataflow 4.9.0 available.

Then, when you run the unit tests, it throws this exception:

```
System.IO.FileLoadException : Could not load file or assembly 'System.Threading.Tasks.Dataflow, Version=4.6.2.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' or one of its dependencies. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040)
   at System.Signature.GetSignature(Void* pCorSig, Int32 cCorSig, RuntimeFieldHandleInternal fieldHandle, IRuntimeMethodInfo methodHandle, RuntimeType declaringType)
   at System.Reflection.RuntimeMethodInfo.FetchNonReturnParameters()
   at System.Reflection.RuntimeMethodInfo.GetParameters()
   at Castle.DynamicProxy.Generators.MethodSignatureComparer.GetHashCode(MethodInfo obj)
   at System.Linq.Set`1.InternalGetHashCode(TElement value)
   at System.Linq.Set`1.Find(TElement value, Boolean add)
   at System.Linq.Enumerable.<DistinctIterator>d__64`1.MoveNext()
   at System.Linq.Buffer`1..ctor(IEnumerable`1 source)
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at Castle.DynamicProxy.Generators.MethodFinder.GetAllInstanceMethods(Type type, BindingFlags flags)
   at Castle.DynamicProxy.Contributors.MembersCollector.CollectMethods(IProxyGenerationHook hook)
   at Castle.DynamicProxy.Contributors.MembersCollector.CollectMembersToProxy(IProxyGenerationHook hook)
   at Castle.DynamicProxy.Contributors.InterfaceProxyWithoutTargetContributor.<CollectElementsToProxyInternal>d__3.MoveNext()
   at Castle.DynamicProxy.Contributors.CompositeTypeContributor.CollectElementsToProxy(IProxyGenerationHook hook, MetaType model)
   at Castle.DynamicProxy.Generators.InterfaceProxyWithoutTargetGenerator.GenerateType(String typeName, Type proxyTargetType, Type[] interfaces, INamingScope namingScope)
   at Castle.DynamicProxy.Generators.InterfaceProxyWithTargetGenerator.<>c__DisplayClass6_0.<GenerateCode>b__0(String n, INamingScope s)
   at Castle.DynamicProxy.Generators.BaseProxyGenerator.ObtainProxyType(CacheKey cacheKey, Func`3 factory)
   at Castle.DynamicProxy.ProxyGenerator.CreateInterfaceProxyWithoutTarget(Type interfaceToProxy, Type[] additionalInterfacesToProxy, ProxyGenerationOptions options, IInterceptor[] interceptors)
   at Moq.Mock`1.InitializeInstancePexProtected() in C:\projects\moq4\src\Moq\Mock.Generic.cs:line 264
   at Moq.Mock`1.OnGetObject() in C:\projects\moq4\src\Moq\Mock.Generic.cs:line 277
   at Moq.Mock`1.get_Object() in C:\projects\moq4\src\Moq\Mock.Generic.cs:line 208
   at ClassLibrary2.Class1.Reproduce() in D:\Development\uaclientdataflow\ClassLibrary2\ClassLibrary2\Class1.cs:line 16

```

basprins, 2019-07-02:
If it doesn't reproduce, could you check if you have dataflow 4.8.0 installed in your gac? 

I just tried the following:
- run unit test with dataflow 4.9.0 referenced in class library1 => FileLoadException
- remove explicit dataflow 4.9.0 package from class library 1, rerun unit test (dataflow 4.8.0 is installed via nuget) => test succeeds
- re-added explicit package ref to dataflow 4.9.0 => FileLoadException
- installed dataflow 4.8.0 into gac => test succeeds
- delete dataflow 4.8.0 from gac => FileLoadException

I hope you can have another close look at this.

Many thanks in advance

awcullen, 2019-07-04:
Sorry about closing this issue. I have upgraded the Dataflow package to 4.9.  
Look for the version 2.3.0 in Nuget soon.

basprins, 2019-07-04:
Hey Andrew,
Thanks for that! I think the root cause of the issue actually is in Moq, still have no idea what it is. But this will fix my problem! Thx a lot!

closed 2019-07-04 by basprins