convertersystems/opc-ua-client #95: [Question] Browser does not find my InstanceDBs in DataBlocksInstance

dojo90, 2019-10-07:
If I use the following code sample, I get all items except my in `DataBlocksInstance`:

```csharpprivate static async Task TestAsync()
{
	var discoveryUrl = "opc.tcp://10.10.1.201:4840"; // UaCppServer - see  http://www.unified-automation.com/
	// var discoveryUrl = $"opc.tcp://localhost:26543"; // Workstation.RobotServer

	Console.WriteLine("Step 1 - Describe this app.");
	var appDescription = new ApplicationDescription()
	{
		ApplicationName = "MyHomework",
		ApplicationUri = $"urn:10.10.1.201:4840/EA17-00017:Test-CPU1500",
		ApplicationType = ApplicationType.Client,
	};

	Console.WriteLine("Step 2 - Create a certificate store.");
	var certificateStore = new DirectoryStore(
		Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData), "Workstation.ConsoleApp", "pki"));

	Console.WriteLine("Step 3 - Create a session with your server.");
	var channel = new UaTcpSessionChannel(
		appDescription,
		certificateStore,
		ShowSignInDialog,
		discoveryUrl);
	try
	{
		await channel.OpenAsync();

		Console.WriteLine($"  Opened channel with endpoint '{channel.RemoteEndpoint.EndpointUrl}'.");
		Console.WriteLine($"  SecurityPolicyUri: '{channel.RemoteEndpoint.SecurityPolicyUri}'.");
		Console.WriteLine($"  SecurityMode: '{channel.RemoteEndpoint.SecurityMode}'.");
		Console.WriteLine($"  UserIdentity: '{channel.UserIdentity}'.");

		//Console.WriteLine("Press any key to continue...");
		//Console.ReadKey(true);

		Console.WriteLine("Step 4 - Browse the server namespace.");
		Console.WriteLine("+ Root");
		BrowseRequest browseRequest = new BrowseRequest
		{
			NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = NodeId.Parse(ObjectIds.RootFolder), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
		};
		BrowseResponse browseResponse = await channel.BrowseAsync(browseRequest);
		foreach (var rd1 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
		{
			Console.WriteLine("  + {0}: {1}, {2}", rd1.DisplayName, rd1.BrowseName, rd1.NodeClass);
			browseRequest = new BrowseRequest
			{
				NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd1.NodeId, channel.NamespaceUris), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
			};
			browseResponse = await channel.BrowseAsync(browseRequest);
			foreach (var rd2 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
			{
				Console.WriteLine("    + {0}: {1}, {2}", rd2.DisplayName, rd2.BrowseName, rd2.NodeClass);
				browseRequest = new BrowseRequest
				{
					NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd2.NodeId, channel.NamespaceUris), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
				};
				browseResponse = await channel.BrowseAsync(browseRequest);
				foreach (var rd3 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
				{
					Console.WriteLine("      + {0}: {1}, {2}", rd3.DisplayName, rd3.BrowseName, rd3.NodeClass);

					if (rd3.BrowseName.Name == "DataBlocksInstance")
					{
						browseRequest = new BrowseRequest
						{
							NodesToBrowse = new BrowseDescription[] { new BrowseDescription { NodeId = ExpandedNodeId.ToNodeId(rd3.NodeId, channel.NamespaceUris), BrowseDirection = BrowseDirection.Forward, ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences), NodeClassMask = (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, IncludeSubtypes = true, ResultMask = (uint)BrowseResultMask.All } },
						};
						browseResponse = await channel.BrowseAsync(browseRequest);

						foreach (var rd4 in browseResponse.Results[0].References ?? new ReferenceDescription[0])
						{
							Console.WriteLine("        + {0}: {1}, {2}", rd4.DisplayName, rd4.BrowseName, rd4.NodeClass);

						}
					}
				}
			}
		}

		Console.WriteLine("Press any key to continue...");
		Console.ReadKey(true);

		Console.WriteLine("Step 5 - Create a subscription.");
		var subscriptionRequest = new CreateSubscriptionRequest
		{
			RequestedPublishingInterval = 1000,
			RequestedMaxKeepAliveCount = 10,
			RequestedLifetimeCount = 30,
			PublishingEnabled = true
		};
		var subscriptionResponse = await channel.CreateSubscriptionAsync(subscriptionRequest);
		var id = subscriptionResponse.SubscriptionId;

		Console.WriteLine("Step 6 - Add items to the subscription.");
		var itemsToCreate = new MonitoredItemCreateRequest[]
		{
			new MonitoredItemCreateRequest { ItemToMonitor = new ReadValueId { NodeId = NodeId.Parse("i=2258"), AttributeId = AttributeIds.Value }, MonitoringMode = MonitoringMode.Reporting, RequestedParameters = new MonitoringParameters { ClientHandle = 12345, SamplingInterval = -1, QueueSize = 0, DiscardOldest = true } }
		};
		var itemsRequest = new CreateMonitoredItemsRequest
		{
			SubscriptionId = id,
			ItemsToCreate = itemsToCreate,
		};
		var itemsResponse = await channel.CreateMonitoredItemsAsync(itemsRequest);

		Console.WriteLine("Step 7 - Subscribe to PublishResponse stream.");
		var token = channel.Where(pr => pr.SubscriptionId == id).Subscribe(pr =>
		{
			// loop thru all the data change notifications
			var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
			foreach (var dcn in dcns)
			{
				foreach (var min in dcn.MonitoredItems)
				{
					Console.WriteLine($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
				}
			}
		});

		Console.WriteLine("Press any key to delete the subscription...");
		while (!Console.KeyAvailable)
		{
			await Task.Delay(500);
		}

		Console.ReadKey(true);

		Console.WriteLine("Step 8 - Delete the subscription.");
		var request = new DeleteSubscriptionsRequest
		{
			SubscriptionIds = new uint[] { id }
		};
		await channel.DeleteSubscriptionsAsync(request);
		token.Dispose();

		Console.WriteLine("Press any key to close the session...");
		Console.ReadKey(true);

		Console.WriteLine("Step 9 - Close the session.");
		await channel.CloseAsync();
	}
	catch (ServiceResultException ex)
	{
		if ((uint)ex.HResult == StatusCodes.BadSecurityChecksFailed)
		{
			Console.WriteLine("Error connecting to endpoint. Did the server reject our certificate?");
		}

		await channel.AbortAsync();
		throw;
	}
}
```
[image]

If I use the sample application from OPCFoundation (https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Samples/Client.Net4) I can see (and watch/subscribe) them.

[image]

Can someone please help me what I'm doing wrong?! 🤔

dojo90, 2019-10-07:
I found it out. Had to browser deeper/use the `BrowseNextAsync()`:

```csharp
public static async Task<List<ReferenceDescription>> BrowseAsync(this UaTcpSessionChannel channel, ExpandedNodeId nodeId)
{
	var browseRequest = new BrowseRequest
	{
		NodesToBrowse = new BrowseDescription[]
		{
			new BrowseDescription
			{
				NodeId = ExpandedNodeId.ToNodeId(nodeId, channel.NamespaceUris),
				BrowseDirection = BrowseDirection.Forward,
				ReferenceTypeId = NodeId.Parse(ReferenceTypeIds.HierarchicalReferences),
				NodeClassMask = 0u,
				IncludeSubtypes = true, ResultMask = (uint) BrowseResultMask.All
			}
		},
	};
	var browseResponse = await channel.BrowseAsync(browseRequest);

	var continuationPoints = browseResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null).ToArray();
	var rds = new List<ReferenceDescription>(browseResponse.Results[0].References ?? new ReferenceDescription[0]);
	while (continuationPoints.Length > 0)
	{
		var browseNextRequest = new BrowseNextRequest { ContinuationPoints = continuationPoints, ReleaseContinuationPoints = false };
		var browseNextResponse = await channel.BrowseNextAsync(browseNextRequest);
		rds.AddRange(browseNextResponse.Results.Where(result => result.References != null).SelectMany(result => result.References));
		continuationPoints = browseNextResponse.Results.Select(br => br.ContinuationPoint).Where(cp => cp != null).ToArray();
	}

	return rds;
}
```

closed 2019-10-07 by dojo90

awcullen, 2019-10-07:
Well done! I'm glad you found your answer.