convertersystems/opc-ua-client #96: GetValueOrDefault<Foo>() is always null

dojo90, 2019-10-07:
I have a `Foo struct` on my Siemens PLC and used in an instance dataBlock:

[image]

Now I would like to read the value and convert it into a `Structure` - `Foo`.

```csharp
var channel = new UaTcpSessionChannel(
	appDescription,
	certificateStore,
	ShowSignInDialog,
	discoveryUrl,
	additionalTypes: new []{typeof(Foo)});

....

ReadRequest readRequest = new ReadRequest
{
	NodesToRead = new[]
	{
		new ReadValueId
		{
			NodeId = zlsStatic[0].NodeId.NodeId,
			AttributeId = AttributeIds.Value
		},
	}
};
var readResult = await channel.ReadAsync(readRequest);
var foo = readResult.Results[0].GetValueOrDefault<Foo>();
```
---
```csharp
[BinaryEncodingId("nsu=http://www.siemens.com/simatic-s7-opcua;s=TE_Foo")]
[DataTypeId("nsu=http://www.siemens.com/simatic-s7-opcua;s=TE_Foo")]
public class Foo : Structure
{
	public bool Act { get; set; }

	public bool Rep { get; set; }

	public override void Encode(IEncoder encoder)
	{
		encoder.WriteBoolean("ACT", Act);
		encoder.WriteBoolean("REP", Rep);
	}

	public override void Decode(IDecoder decoder)
	{
		Act = decoder.ReadBoolean("ACT");
		Rep = decoder.ReadBoolean("REP");
	}
}
```

[image]

But `foo` is always `null` and the `Decode()` is never invoked.

Am I missing something? 🤔

awcullen, 2019-10-07:
Two things.
You will find that Siemens prefers to add quotes around symbols, like "Foo".  So the

awcullen, 2019-10-07:
Second thing is that the BinaryEncodingId and DataTypeId will be different.

dojo90, 2019-10-07:
As you are writing this answer, I also saw my mistake of the missing `"`. 😊👍

Is there a special reason why `BinaryEncodingIdAttribute` is `sealed`? Can you remove it, that I could create a (for example) `"SiemensStructAttribute"` where I only have to insert my struct name?

Where do I need `DataTypeId` and where do I get the information it's value? Actually it is very hard to get into the "OPC UA" libraries because of missing documentation, but btw. your library is MUCH more better then the one from `OPC Foundation`, as you also implemented `async/await` 👌 and `observables` 💖

awcullen, 2019-10-07:
I guess the DataTypeIdAttribute is unused.  Lol.
I am working on adding a feature to my tool UaBrowser,  a free Visual Studio Extension in the VS Marketplace,  that would allow you to drag a DataType from a tree and drop it into your code, producing the code snippet like you have above.

dojo90, 2019-10-07:
Is there a special reason why `BinaryEncodingIdAttribute` is sealed? Can you remove it, that I could create a (for example) `"SiemensStructAttribute"` where I only have to insert my struct name?

closed 2019-10-08 by dojo90